don/cordova-plugin-ble-central #463: Working in android but not in IOS

deligence, 2017-09-29:
Hello

I am using this plugin and testing on both android and IOS, I have a app which uses BLE device to communicate, We are tracking stats of this BLE device and showing in our mobile devices, everything seems to be working fine on both the platforms except one thing, which is when we stop our BLE device it should not increment timer in our application, but it is not working in IOS, but working fine with android, also this issue only occurs when our application is running in background or when the mobile screen is shut off(working fine in foreground), I have research but not found anything related to my requirement.

I am using iPhone 6s with IOS 11 and using latest version of this plugin i.e. 1.1.4, also check on 1.1.3, I am creating application in meteor 1.3.

Can you please help me out with this issue?

Thanks

Math3v, 2017-10-07:
@deligence Same here!

ddugue, 2017-10-20:
I'm trying to understand your issue and how it is related to cordova-plugin-ble-central.

You want to receive notifications when the app is paused (or in background) and you are *not* receiving notifications 
OR 
You DON'T want to receive notifications when the app is paused (or in background) and you ARE receiving notifications

closed 2018-06-07 by don

subratpalhar92, 2020-09-07:
make sure you are using write() instead of writeWithoutResponse(), this solved my issue on ios