don/cordova-plugin-ble-central #464: Device stays connected on app close/exit

nimeso, 2017-09-30:
We have a Bluetooth button which starts advertising when you click the button once.
In the software we have a button you click to scan/connect to the device and another button you click to disconnect the device. When the device is connected it is no longer advertising.

This is all working great BUT if a user force closes/exits the app while the button is connected it means that the button is not advertising and therefore will not show up if the user reopens the app and scans.

What we need is like a onApplicationExit event so I could disconnect but from what I'm readying there is no such event. We can't use the window pause/resume events because we need the device to keep running if the app goes in to the background.

I know the nRF app, disconnects all devices on app exit/close. Maybe you guys have got some ideas?

Cheers
J

ddugue, 2017-10-11:
Normally, when you force close the app on android, it should close all the BLE connection and garbage collect them automatically.

Although, some phones have terrible BLE implementation and their connection never closes properly!!! Manually turning the bluetooth solution on/off is a way to go.

We could try to hook disconnect calls from the onDestroy activity handler, the problem is since the calls for disconnection are async in nature, we would not even be sure that it cleanly disconnects.

If you are able to change your BLE dongle firmware, you could implement a ping routine every min or so and if no ping, it disconnects the dongle.

don, 2017-10-23:
Like @ddugue says, killing the app should kill the connection. So maybe something is in a bad state on the peripheral? Can you change the firmware on the peripheral to start advertising when the central device disconnects?

For my Bluetooth workshops with PhoneGap I use a `window.beforeunload` hook to disconnect the  peripheral. Something similar might work for you. https://github.com/don/phonegap-ble-workshop/blob/master/template/template_src/www/js/disconnector.js#L73-L79

closed 2018-06-07 by don