don/cordova-plugin-ble-central #469: readRSSI hangs on Android

K1N5L4Y3R, 2017-10-05:
After a quick search on this repository's issues, I found that in https://github.com/don/cordova-plugin-ble-central/issues/313, @mebrunet mentions that he had problems while reading the RSSI values on Android devices. I can confirm that I also have this issues, so I'm opening this separate issue (since this topic doesn't appear to be what #313 is about).

I am having issues calling `readRSSI()` inside an interval that runs every n milliseconds on Android. After some time (maybe a minute or two) of connecting to two BLE peripherals, they hang, meaning they don't return new RSSI values, even though the interval is running. In order to refresh their values, I need to disconnect and reconnect to the peripheral. I have seen the issue on my Moto G5 Plus running Android 7.0. I can confirm this only occurs with Android, since I have the same codebase running in an iPhone 7 that was running iOS 10.x and now 11.0.1 and 11.0.2, and I haven't observed the same issue there.

Do you think this has to do with connecting to `two` peripherals simultaneously? Any ideas to overcome this situation, since our app will highly depend on having RSSI values constantly updated?

ddugue, 2017-10-11:
Hey,
I'm working extensively with bluetooth ble device and this plugin. 
Conclusion: Android is pretty terrible at handling multiple BLE devices connection at the same time.
It is not an issue with android _per se_, but with the way the different manufacturers handle the implementation of the android firmware. So sometimes, it works with one type of phone and it works with another. First step would be to try with a different phone (which has a good BLE support) to see if it is related to the library.

When I was testing, I've noticed a bug on android where the polling queue of commands was not emptied on disconnection, it _might_ be related, I'll try to submit a pull request.

don, 2017-10-23:
> Conclusion: Android is pretty terrible at handling multiple BLE devices connection at the same time.
That's unfortunate but not surprising. That would be excellent one of you is able to track down a work around to this.

ddugue, 2017-10-23:
My pull request #481 can improve disconnection a bit.

What we ended up doing in our app is to put as much information as possible in the broadcasting data and we try to avoid multiple BLE connections at the same time.
We also implemented a ping mechanism where without any ping after ~1 min, the BLE device would disconnect.
We also noticed, that it depends largely on the phone model (more than the android version), so you could try to see if it works with other android phones.

Good luck,

K1N5L4Y3R, 2017-10-23:
Thanks for your input. We will test to see if the changes introduced by your PR help with our issue.

We are aware that there is a high dependance on each Android devices capabilities, and that things can change from one device to another, but any efforts to improve general reliability are highly appreciated :)

tempestwf, 2018-03-23:
I am having this issue too. Was there ever a resolution?

closed 2018-06-07 by don