don/cordova-plugin-ble-central #472: Any issues on iOS 11?

qpwr, 2017-10-09:
This is not an issue, but a question: are there any known issues with this plugin in iOS 11 (e.g. connection drops etc.)? I am currently using the evothings plugin which has been randomly discontinued and would like to transition over to a new plugin because the evothings stuff doesn't work properly on iOS 11.

Thanks for any input on this.

ccravens, 2017-10-10:
If BLE is turned off with iOS 11, enabling does not invoke iOS 11 to ask if the user if BLE should be enabled.

don, 2017-10-23:
@qpwr it's software there are always issues. Generally though this works really well on iOS and on newer Android devices. The API is pretty consistent across iOS and Android.

closed 2017-10-23 by don

mrcbnsls, 2017-10-24:
@don, thanks for the update. I was actually referring (but not clear enough, my fault) to a specific issue that many people are having right now with BLE on iOS 11 (as described above). But if nothing like this is known right now, this may be a good sign.