don/cordova-plugin-ble-central #474: iOS 11 does not ask user to enable bluetooth

ccravens, 2017-10-10:
In iOS 11, if bluetooth is disabled, iOS does not ask the user if they would like to enable bluetooth when running "enable" as with previous versions of iOS.

ccravens, 2017-10-10:
Apparently if Bluetooth is disabled via the "Control Center" (swipe up from the bottom) doesn't actually disable bluetooth, it just turns it off for everything except for Apple's products. If you disable Bluetooth from the Settings app, then it will ask to enable.

http://www.telegraph.co.uk/technology/2017/09/21/ios-11s-bluetooth-wi-fi-controls-dont-actually-turn-bluetooth/

luposlip, 2017-10-14:
It works for me via the Control Center (buttom swipe-up menu). On iOS 11.0.2 on iPhone SE.

Check the extra states for iOS (and Android for that matter) (`resetting`, `unknown` etc.):
https://github.com/don/cordova-plugin-ble-central/#description-13

I've set my app state to `off` if not explicitly `on` or `resetting`.

don, 2017-10-23:
@ccravens I think you're referring to the automatic iOS dialog (See #174.) For iOS 11 I think you need to do check for the Bluetooth state yourself like @luposlip mentions using [ble.startStateNotifications](https://github.com/don/cordova-plugin-ble-central/#startstatenotifications)

ccravens, 2017-10-26:
Thanks @don we're looking into this and I'll update the status of this issue as soon as can figure it out and post details.

don, 2018-06-07:
@ccravens if ble.enable was the first call you made to the plugin, that would initialize the CBCentralManager, which prompted the user to enable Bluetooth. So this was just a happy accident that it did what you wanted.

https://github.com/don/cordova-plugin-ble-central/blob/8a8b43ceefe3946a87727304baadc173d93c6357/src/ios/BLECentralPlugin.m#L41

Pull request #584 removes that prompt. ble.enable doesn't work on iOS but version 1.2.0 added ble.showBluetoothSettings

closed 2018-06-07 by don