don/cordova-plugin-ble-central #475: invocationTargetException when requesting permissions [android.permission.ACCESS_COARSE_LOCATION]

wzhsh90, 2017-10-10:
mobile model: HUAWEI GRA-UL10 
os: android 6.0

ble start scan need location permission, but I still click forbidden,after server times ,the request of permission make the app crash.

1、 Can reqeust only one set of permissions at a time (log many times)
2、 java.lang.reflect.InvocationTargetException
                                                                                at java.lang.reflect.Method.invoke(Native Method)
                                                                                at org.apache.cordova.PermissionHelper.requestPermissions(PermissionHelper.java:65)
                                                                                at org.apache.cordova.PermissionHelper.requestPermission(PermissionHelper.java:45)
                                                                                at com.megster.cordova.ble.central.BLECentralPlugin.findLowEnergyDevices(BLECentralPlugin.java:463)
                                                                                at com.megster.cordova.ble.central.BLECentralPlugin.onRequestPermissionResult(BLECentralPlugin.java:591)
                                                                                at org.apache.cordova.CordovaInterfaceImpl.onRequestPermissionResult(CordovaInterfaceImpl.java:214)
                                                                                at org.apache.cordova.CordovaActivity.onRequestPermissionsResult(CordovaActivity.java:508)
                                                                                at android.app.Activity.requestPermissions(Activity.java:3917)
                                                                                at org.apache.cordova.CordovaInterfaceImpl.requestPermissions(CordovaInterfaceImpl.java:226)
                                                                                at java.lang.reflect.Method.invoke(Native Method)
                                                                                at org.apache.cordova.PermissionHelper.requestPermissions(PermissionHelper.java:65)
                                                                                at org.apache.cordova.PermissionHelper.requestPermission(PermissionHelper.java:45)
                                                                                at com.megster.cordova.ble.central.BLECentralPlugin.findLowEnergyDevices(BLECentralPlugin.java:463)
                                                                                at com.megster.cordova.ble.central.BLECentralPlugin.onRequestPermissionResult(BLECentralPlugin.java:591)
                                                                                at org.apache.cordova.CordovaInterfaceImpl.onRequestPermissionResult(CordovaInterfaceImpl.java:214)
                                                                                at org.apache.cordova.CordovaActivity.onRequestPermissionsResult(CordovaActivity.java:508)
                                                                                at android.app.Activity.requestPermissions(Activity.java:3917)
                                                                                at org.apache.cordova.CordovaInterfaceImpl.requestPermissions(CordovaInterfaceImpl.java:226)
                                                                                at java.lang.reflect.Method.invoke(Native Method)
                                                                                at org.apache.cordova.PermissionHelper.requestPermissions(PermissionHelper.java:65)
                                                                                at org.apache.cordova.PermissionHelper.requestPermission(PermissionHelper.java:45)
                                                                                at com.megster.cordova.ble.central.BLECentralPlugin.findLowEnergyDevices(BLECentralPlugin.java:463)
                                                                                at com.megster.cordova.ble.central.BLECentralPlugin.onRequestPermissionResult(BLECentralPlugin.java:591)
                                                                                at org.apache.cordova.CordovaInterfaceImpl.onRequestPermissionResult(CordovaInterfaceImpl.java:214)
                                                                                at org.apache.cordova.CordovaActivity.onRequestPermissionsResult(CordovaActivity.java:508)
                                                                                at android.app.Activity.requestPermissions(Activity.java:3917)
                                                                                at org.apache.cordova.CordovaInterfaceImpl.requestPermissions(CordovaInterfaceImpl.java:226)
                                                                                at java.lang.reflect.Method.invoke(Native Method)
                                                                                at org.apache.cordova.PermissionHelper.requestPermissions(PermissionHelper.java:65)
                                                                                at org.apache.cordova.PermissionHelper.requestPermission(PermissionHelper.java:45)
                                                                                at com.megster.cordova.ble.central.BLECentralPlugin.findLowEnergyDevices(BLECentralPlugin.java:463)
                                                                                at com.megster.cordova.ble.central.BLECentralPlugin.onRequestPermissionResult(BLECentralPlugin.java:591)
                                                                                at org.apache.cordova.CordovaInterfaceImpl.onRequestPermissionResult(CordovaInterfaceImpl.java:214)
                                                                                at org.apache.cordova.CordovaActivity.onRequestPermissionsResult(CordovaActivity.java:508)
                                                                                at android.app.Activity.requestPermissions(Activity.java:3917)
                                                                                at org.apache.cordova.CordovaInterfaceImpl.requestPermissions(CordovaInterfaceImpl.java:226)
                                                                                at java.lang.reflect.Method.invoke(Native Method)
                                                                                at org.apache.cordova.PermissionHelper.requestPermissions(PermissionHelper.java:65)
                                                                                at org.apache.cordova.PermissionHelper.requestPermission(PermissionHelper.java:45)
                                                                                at com.megster.cordova.ble.central.BLECentralPlugin.findLowEnergyDevices(BLECentralPlugin.java:463)
                                                                                at com.megster.cordova.ble.central.BLECentralPlugin.onRequestPermissionResult(BLECentralPlugin.java:591)
                                                                                at org.apache.cordova.CordovaInterfaceImpl.onRequestPermissionResult(CordovaInterfaceImpl.java:214)
                                                                                at org.apache.cordova.CordovaActivity.onRequestPermissionsResult(CordovaActivity.java:508)
                                                                                at android.app.Activity.requestPermissions(Activity.java:3917)
                                                                                at org.apache.cordova.CordovaInterfaceImpl.requestPermissions(CordovaInterfaceImpl.java:226)
                                                                            	at java.lang.refl

don, 2018-06-08:
I'm not sure what's causing this. Closing this since it's an old issue. Open a new issue if it's still a problem.

closed 2018-06-08 by don

wzhsh90, 2018-06-08:
<dependency id="cordova-plugin-compat" version="^1.2.0" />

 this dependency is no need anymore, the cordova default contains it ,why not remove it ?