don/cordova-plugin-ble-central #476: Re-enabling notifications after a stopNotifications fail on iOS doesn't work

ddugue, 2017-10-11:
In order to reproduce this bug:
- Call startNotification on a peripheral
- Call stopNotification on the same peripheral
- Call startNotification on the same peripheral
- Notice that the callback never happens

Happened on Iphone 5 with iOS 10.1.3

Also, I've notices that the callback that I sent with stopNotification gets called when re-enabling the notification. Please note that it is NOT called when calling stopNotification.

It is probably linked to the code in didUpdateNotificationStateForCharacteristic:
since the stopNotificationCallback is not called, it calls the stopNotificationCallback WHEN reconnecting and thus removes the notification callback from the list.

ddugue, 2017-10-20:
I have been able to fix this issue with pull request
see #482

zabojad, 2018-05-25:
I'm probably having the same issue. I'll test your fix in a moment.

commit ac55995d (2018-06-13): Call stopNotification callback at the right time
Closes #476 and closes #482 Thanks @ddugue
closed 2018-06-13 by don