don/cordova-plugin-ble-central #477: Not consistent connection

Thyrion33, 2017-10-13:
Hello,

I use your plugin,  for an application that command a motorized door, and allow user to configure it.
Before using the app, the user have to bond the door with the smartphone.

The problem that i have is that most of the time, the connection to the door work without any problem and is fast.
But sometimes it take longer (like 2-3 seconds).
And sometimes the connection just fail. After a fail most of the time the next attempt will fail too.

I don't know what happend and i don't how i could have more informations about this not consistant problem inside ionic.

Other thing which i doesn't know if it is related or not, is that after a disconnection, sometimes, the smartphone doesn't see the door during the scan (however the ble chip is advertising...)

The bluetooth chip that i use is a Cypress BLE ProC chip and i have tested that side it with a developpment usb key to "simulate" the phone and everythings is seems OK on this side, connection is always fast and never fail, the device is always detected after a a disconnection when we scan.

Thanks for your help.

don, 2017-10-23:
I suggest enabling the [Bluetooth HCI snoop log](https://blog.bluetooth.com/debugging-bluetooth-with-an-android-app) on your phone. When you see a connection problem, load the log into wireshark and see what happened.

Sometimes service and characteristic discovery takes a while. Bonding should avoid this, but you could edit the Java code to only discover things you care about.

Sometimes on some devices the Bluetooth stack crashes, you should see something in `adb logcat`.

Thyrion33, 2017-10-24:
Thanks for you answers.
From adb logcat it seems that the scanning problem appear : 
"E BtGatt.GattService: App 'com.blabla.mlpcUser' is scanning too frequently", and seems related too : "We’ve changed the BLE Scanning behavior starting in DP4. We’ll prevent applications from starting and stopping scans more than 5 times in 30 seconds. For long running scans, we’ll convert them into opportunistic scans."
I have it because i'm testing the APP by making connect/disconnect cycles. It's not related to your plugin.
What is  the common way to handle that ? Prevent user like "you scan too frequenlty, wait x seconds before a new scan attempt?"

For the connection problem i don't see anything on the log file

ddugue, 2017-10-30:
Hey @Thyrion33,

For scanning, I guess you could disable your user button for ~5 seconds on stopScan and ~2 seconds on startScan, it would then not trigger the error you've received.

For the connection error, do you receive 133 error? You could verify that via adb logcat

Thyrion33, 2017-11-06:
Hi, thanks for your answers dduge

For the disconnection error from HCI log, it is a  0x03e error :

[image]

ddugue, 2017-11-08:
I *think* the full list of bluetooth status can be found here:
[https://devzone.nordicsemi.com/documentation/nrf51/4.1.0/html/group___b_l_e___h_c_i___s_t_a_t_u_s___c_o_d_e_s.html](https://devzone.nordicsemi.com/documentation/nrf51/4.1.0/html/group___b_l_e___h_c_i___s_t_a_t_u_s___c_o_d_e_s.html)

What about adb logcat, do you receive any other errors that the "Scanning too frequently"?

Thyrion33, 2017-11-20:
Sorry for the long delay..

Here the ADB logcat when the error happend :

11-20 10:46:35.221  2609  2683 D BLEPlugin: action = connect
11-20 10:46:35.226  2609  2683 D BluetoothGatt: connect() - device: 00:A0:50:1D:63:XX, auto: false
11-20 10:46:35.226  2609  2683 D BluetoothAdapter: isSecureModeEnabled
11-20 10:46:35.226  2211  2224 D BtConfig.SecureMode: isSecureModeOn:false
11-20 10:46:35.226  2609  2683 D BluetoothGatt: registerApp()
11-20 10:46:35.226  2609  2683 D BluetoothGatt: registerApp() - UUID=fd50adf9-881b-41ff-9901-3998931e9595
11-20 10:46:35.227  2211  2848 D BtGatt.GattService: registerClient() - UUID=fd50adf9-881b-41ff-9901-3998931e9595
11-20 10:46:35.246  2759  2999 D lights  : button : 0 +
11-20 10:46:35.252 28873 28873 I wpa_supplicant: Scan requested (ret=0) - scan timeout 30 seconds
11-20 10:46:35.263  2759  5496 D MountService: getExternalStorageMountMode : 1
11-20 10:46:35.264  2759  5496 D MountService: getExternalStorageMountMode : 3
11-20 10:46:35.264  2759  5496 D MountService: getExternalStorageMountMode : final mountMode=1, uid : 1000, packageName : com.samsung.android.lool
11-20 10:46:35.274  2759  2999 D lights  : button : 0 -
11-20 10:46:35.277  5048  5048 E Zygote  : v2
11-20 10:46:35.277  5048  5048 I libpersona: KNOX_SDCARD checking this for 1000
11-20 10:46:35.277  5048  5048 I libpersona: KNOX_SDCARD not a persona
11-20 10:46:35.278  5048  5048 E Zygote  : accessInfo : 0
11-20 10:46:35.278  5048  5048 W SELinux : SELinux selinux_android_compute_policy_index : Policy Index[2],  Con:u:r:zygote:s0 RAM:SEPF_SECMOBILE_7.0_0009, [-1 -1 -4 -1 0 1]
11-20 10:46:35.280  5048  5048 I SELinux : SELinux: seapp_context_lookup: seinfo=platform, pkgname=com.samsung.android.lool 
11-20 10:46:35.282  2759  5496 I ActivityManager: Start proc 5048:com.samsung.android.lool/1000 for broadcast com.samsung.android.lool/com.samsung.android.sm.ui.battery.AppLockingUserReceiver
11-20 10:46:35.307  5048  5048 D TimaKeyStoreProvider: TimaKeyStore is not enabled: cannot add TimaSignature Service and generateKeyPair Service
11-20 10:46:35.310  2759  5494 I ActivityManager: DSS on for com.samsung.android.lool and scale is 1.0
11-20 10:46:35.323  2759  5498 D CustomFrequencyManagerService: releaseDVFSLockLocked : Getting Lock type frm List : DVFS_MIN_LIMIT  frequency : 700000  uid : 1000  pid : 2759  tag : UNLOCKSCREEN_BOOSTER@2
11-20 10:46:35.329  2609  2683 W PluginManager: THREAD WARNING: exec() call to BLE.connect blocked the main thread for 108ms. Plugin should use CordovaInterface.getThreadPool().
11-20 10:46:35.330  2211  2233 D BtGatt.GattService: onClientRegistered() - UUID=fd50adf9-881b-41ff-9901-3998931e9595, clientIf=7, status=0
11-20 10:46:35.331  2609  2622 D BluetoothGatt: onClientRegistered() - status=0 clientIf=7
11-20 10:46:35.337  2211  2847 D BtGatt.GattService: clientConnect(com.wimove.mlpcUser) - address = 00:A0:50:1D:63:XX, isDirect=true transport =2 set own addr = false own addr type:0, clientIf: 7
11-20 10:46:35.342  2211  2233 W bt_btif : btif_dm_generic_evt: event=33037
11-20 10:46:35.343  5048  5048 I art     : Starting a blocking GC AddRemoveAppImageSpace
11-20 10:46:35.421  2609  2609 D InputMethodManager: HSI from window - flag : 0 Pid : 2609
11-20 10:46:35.421  2759  2909 V InputMethodManagerService: Client requesting input be hidden, flags : 0
11-20 10:46:35.421  2759  2909 V InputMethodManagerService: hideCurrentInputLocked - !shouldHideSoftInput
11-20 10:46:35.444  5048  5048 D SamsungAnalytics:1.8.32: cf feature is supported
11-20 10:46:35.448  5048  5048 D SamsungAnalytics:1.8.32: [Tracker] Tracker start:1.8.32
11-20 10:46:35.519  2759  3544 D SamsungAlarmManager: setInexact Intent (T:0/F:0/AC:false) 20171120T105235 - CU:10005/CP:4474
11-20 10:46:35.609  6307  6517 D com.facebook.appevents.AppEventsLogger: Got unexpected exception: java.io.WriteAbortedException: writing aborted; java.io.NotSerializableException: org.json.JSONObject
11-20 10:46:35.661  2211  2233 W bt_btif : btif_dm_generic_evt: event=33037
11-20 10:46:35.661  2211  2355 E bt_btm  : btm_ble_read_resolving_list_entry_complete peer_addr: 00:00:00:00:00:00
11-20 10:46:35.680  4556  5012 I System.out: (HTTPLog)-Static: isSBSettingEnabled false
11-20 10:46:35.680  4556  5012 I System.out: (HTTPLog)-Static: isSBSettingEnabled false
11-20 10:46:35.833  4474  5620 I tnet-jni: 203 - [spdyPingRecvCallback] - 
11-20 10:46:35.833  4474  5620 I tnet-jni: 203 - [SpdySessionCallBack.spdyPingRecvCallback] - org.android.spdy.SpdySession@72b1b24
11-20 10:46:35.841  2759  2771 D SamsungAlarmManager: setInexact Intent (T:0/F:0/AC:false) 20171120T105235 - CU:10005/CP:4474
11-20 10:46:35.994  2211  2233 W bt_btif : btif_dm_generic_evt: event=33037
11-20 10:46:35.994  2211  2355 W bt_btif : bta_gattc_conn_cback() - cif=3 connected=0 conn_id=3 reason=0x003e
11-20 10:46:35.994  2211  2355 W bt_btif : bta_gattc_conn_cback() - cif=4 connected=0 conn_id=4 reason=0x003e
11-20 10:46:35.994  2211  2355 W bt_btif : bta_gattc_conn_cback() - cif=5 connected=0 conn_id=5 reason=0x003e
11-20 10:46:35.994  2211  2355 W bt_btif : bta_gattc_conn_cback() - cif=6 connected=0 conn_id=6 reason=0x003e
11-20 10:46:35.994  2211  2355 W bt_btif : bta_gattc_conn_cback() - cif=7 connected=0 conn_id=7 reason=0x003e
11-20 10:46:35.994  2211  2355 W bt_btm  : btm_sec_disconnected
11-20 10:46:35.996  2211  2233 D BtGatt.GattService: onConnected() - clientIf=7, connId=0, address=00:A0:50:1D:63:XX
[... cut ...]
11-20 10:46:36.008  2211  2235 D BtGatt.GattService: clientDisconnect(com.wimove.mlpcUser) - address=00:A0:50:1D:63:XX, connId=null, clientIf: 7
11-20 10:46:36.010  2211  2355 E bt_btif : No such connection need to be cancelled
11-20 10:46:36.010  2609  2685 D BluetoothGatt: close()
11-20 10:46:36.010  2609  2685 D BluetoothGatt: unregisterApp() - mClientIf=7
11-20 10:46:36.012  2211  2236 D BtGatt.GattService: unregisterClient() - clientIf=7
11-20 10:46:36.086  2609  2683 D BLEPlugin: action = disconnect
11-20 10:46:36.199  2609  2683 D BLEPlugin: action = disconnect

don, 2018-01-16:
@Thyrion33 is this still a problem?

Deepankerchaudhary, 2018-01-22:
@Don:
Process of my work:
scanBLE - then connect - write data  then disconnect.
other sequence startBLE - then connect - startNotification - write - disconnect.
Sometime when i try to connect it gives disconnection error in Android 6 RedMI device:
Logs are given belwo:
01-22 21:52:06.207  2519  2660 D BluetoothGatt: connect() - device: F4:3C:3C:0B:32:D7, auto: false

01-22 21:52:06.208  2519  2660 D BluetoothGatt: registerApp()

01-22 21:52:06.208  2519  2660 D BluetoothGatt: registerApp() - UUID=52f7ee6c-9dd2-4c95-97b3-704409256596

01-22 21:52:06.209 29147 29158 D BtGatt.GattService: registerClient() - UUID=52f7ee6c-9dd2-4c95-97b3-704409256596

01-22 21:52:06.209 29147  2682 D BtGatt.GattService: onClientRegistered() - UUID=52f7ee6c-9dd2-4c95-97b3-704409256596, clientIf=7

01-22 21:52:06.210  2519  2605 D BluetoothGatt: onClientRegistered() - status=0 clientIf=7

01-22 21:52:06.210 29147  2779 D A2dpService: getA2DPService(): returning com.android.bluetooth.a2dp.A2dpService@2913432

01-22 21:52:06.211 29147  2779 I A2dpService: audio isMusicActive is false

01-22 21:52:06.211 29147  2779 D BtGatt.GattService: clientConnect() - address=F4:3C:3C:0B:32:D7, isDirect=true

01-22 21:52:06.212 29147  2682 D bt_btif_config: btif_get_address_type: Device [f4:3c:3c:0b:32:d7] address type 1

01-22 21:52:06.212 29147  2682 D bt_btif_config: btif_get_device_type: Device [f4:3c:3c:0b:32:d7] type 2

01-22 21:52:06.366  1447  2934 D BaseMiuiPhoneWindowManager: keyCode:23 down:false 85027822 85027510 policyFlags=22000000 deviceId:0 isScreenOn:true

01-22 21:52:06.502 29147  2690 W bt_btm  : btm_acl_created hci_handle=3 link_role=0  transport=2

01-22 21:52:06.511 29147  2682 E BtGatt.GattService: Exception: android.os.DeadObjectException

01-22 21:52:06.824 29147  2690 W bt_btif : bta_gattc_conn_cback() - cif=3 connected=0 conn_id=3 reason=0x003e

01-22 21:52:06.824 29147  2690 W bt_btif : bta_gattc_conn_cback() - cif=4 connected=0 conn_id=4 reason=0x003e

01-22 21:52:06.824 29147  2690 W bt_btif : bta_gattc_conn_cback() - cif=5 connected=0 conn_id=5 reason=0x003e

01-22 21:52:06.824 29147  2690 W bt_btif : bta_gattc_conn_cback() - cif=6 connected=0 conn_id=6 reason=0x003e

01-22 21:52:06.824 29147  2690 W bt_btif : bta_gattc_conn_cback() - cif=7 connected=0 conn_id=7 reason=0x003e

01-22 21:52:06.826 29147  2682 E BtGatt.GattService: Exception: android.os.DeadObjectException

01-22 21:52:06.826 29147  2690 I bt_btm_sec: btm_sec_disconnected clearing pending flag handle:3 reason:62

01-22 21:52:06.827 29147  2682 D BtGatt.GattService: onConnected() - clientIf=7, connId=0, address=F4:3C:3C:0B:32:D7

01-22 21:52:06.827  2519  2605 D BluetoothGatt: onClientConnectionState() - status=133 clientIf=7 device=F4:3C:3C:0B:32:D7

01-22 21:52:06.830  2519  2605 D BluetoothGatt: cancelOpen() - device: F4:3C:3C:0B:32:D7

01-22 21:52:06.831 29147  2696 D BtGatt.GattService: clientDisconnect() - address=F4:3C:3C:0B:32:D7, connId=null

01-22 21:52:06.831 29147  2690 E bt_btif : No such connection need to be cancelled

01-22 21:52:06.831  2519  2605 D BluetoothGatt: close()

01-22 21:52:06.831  2519  2605 D BluetoothGatt: unregisterApp() - mClientIf=7

01-22 21:52:06.832 29147  2690 E bt_btif : bta_gattc_mark_bg_conn unable to find the bg connection mask for: f4:3c:3c:0b:32:d7

01-22 21:52:06.832 29147  2694 D BtGatt.GattService: unregisterClient() - clientIf=7

01-22 21:52:06.836  2519  2519 D SystemWebChromeClient: file:///android_asset/www/js/services/sync_adharecne_service.js: Line 222 : SyncService:error connecting to device{"name":"MS19_Inhaler","id":"F4:3C:3C:0B:32:D7","errorMessage":"Peripheral Disconnected"}

01-22 21:52:06.836  2519  2519 I chromium: [INFO:CONSOLE(222)] "SyncService:error connecting to device{"name":"MS19_Inhaler","id":"F4:3C:3C:0B:32:D7","errorMessage":"Peripheral Disconnected"}", source: file:///android_asset/www/js/services/sync_adharecne_service.js (222)

[... 24 lines cut ...]
01-22 21:52:08.695 32539 32559 E PulseDataRecorder: 	at X.012.E(:5266)

01-22 21:52:08.695 32539 32559 E PulseDataRecorder: 	at X.0Bz.A(:33106)

01-22 21:52:08.695 32539 32559 E PulseDataRecorder: 	at X.019.B(:5404)

01-22 21:52:08.695 32539 32559 E PulseDataRecorder: 	at X.0DF.run(:36158)

01-22 21:52:08.695 32539 32559 E PulseDataRecorder: Caused by: X.0Dt

01-22 21:52:08.695 32539 32559 E PulseDataRecorder: 	at X.00w.F(:4971)

01-22 21:52:08.695 32539 32559 E PulseDataRecorder: 	... 5 more

01-22 21:52:08.729   807   807 I MSM-irqbalance: Decided to move IRQ155 from CPU0 to CPU2

01-22 21:52:08.737 29147  2682 E BtGatt.GattService: Exception: android.os.DeadObjectException

01-22 21:52:08.944 29147  2682 E BtGatt.GattService: Exception: android.os.DeadObjectException

01-22 21:52:09.055 29147  2682 E BtGatt.GattService: Exception: android.os.DeadObjectException

01-22 21:52:09.176   721   842 W XTCC-5.1.0.13: [CS-RIL-LISTENER] timerCallback fired for ril update registering, re-register

Kindly check & please let me know what's wrong

Thyrion33, 2018-01-29:
@don : 
I still  have the problem, i  tried an other plugin than your and same problem.
What i do temporary is that i try like 5 times to reconnect if connection fail, it's transparent for user, and always work. But i would like to know why sometimes that disconnection during a connection happen. And fix it in a proper way :)

don, 2018-05-07:
@Thyrion33 I'm not sure. These are difficult problems to track down. Maybe you could use [autoConnect](https://github.com/don/cordova-plugin-ble-central#autoconnect) instead of scanning?

don, 2018-06-07:
Closing old issue. Re-open or create a new issue if you have more info in the future.

closed 2018-06-07 by don