don/cordova-plugin-ble-central #479: Can't read from service: "180d" characteristic "2a37" (H7 Polar Heart Rate Monitor)

markodayan, 2017-10-19:
Hi there,

I've been having problems accessing the heart rate readings from the H7 Polar, of which reside within the service UUID "180d" and with characteristic "2a37". My notifications for the heart rate monitor are successful but the reads are always failing with the error "Read Failed". I checked the device and found that this service only has the "notify" property, do you by any chance know how I may get my heartrate data as I havn't been successful with the read so far. Just to also mention I tested the read function on different data within the device and it returned successfully.

Here is the section of the code I am running below, I have just been using console.log for now to determine whether my reads are working for the heart rate information (service: 180d, char:2a37)

```
          this.ble.startNotification(id, '180d','2a38').subscribe( (notificationData) => {
            console.log("Notification:"+ String.fromCharCode.apply(null, new Uint8Array(notificationData)));
            this.ble.read(id, '180d','2a38').then( (data) => {
      
              console.log ("Read" + JSON.stringify(data));
            }, (error) => {
            console.log("Error Read" + JSON.stringify(error));
      
            });
          }, (error) => {
            console.log("Error Notification" + JSON.stringify(error));
          });     
        },
        err => {
          this.isConnected = false;
          this.toast.create({message: `Unsuccessful!`,duration: 3000}).present();
        });  
```

I have attached the error message I receive below from the android console, its fundamentally about why I am getting a reading error for those particular read input parameters

[image]

ddugue, 2017-10-20:
hey @markodayan,

I am not familiar with heart rate information service, but I am aware of how the notifications normally work.

You should not have to call the read function on the characteristic. The data `notificationData` in your example should contain the information you are seeking. You need to look at the documentation of the protocol for your heart rate service.

From what I have gathered from the internet, the bytes sent are either a Uint8 or a Uint16, so you should not transform it in a string (String.fromCharCode).

In any case, this issue doesn't seem to be related to the plugin. I suggest we close it. @don

markodayan, 2017-10-20:
Hi thank you for the reply.

ddugue, 2017-10-23:
@markodayan If you don't convert your data to string, what is the value inside notificationData?

don, 2017-10-23:
@markodayan you should not read inside the notification. The heart rate data is sent to the callback you pass to startNotification.

Characteristic 0x2A37 on Service 0x180D is notify only, it can't be read

[image]

https://www.bluetooth.com/specifications/gatt/viewer?attributeXmlFile=org.bluetooth.service.heart_rate.xml

Once you get the data, you need to decode it using the docs
https://www.bluetooth.com/specifications/gatt/viewer?attributeXmlFile=org.bluetooth.characteristic.heart_rate_measurement.xml

For a simple example heart rate code, I cheat and convert the data to a Uint8Array and grab the 2nd element.
https://github.com/don/cordova-plugin-ble-central/blob/master/examples/heartrate/www/js/index.js#L68-L73

closed 2017-10-23 by don