don/cordova-plugin-ble-central #480: Long connection times

dylanrynhart, 2017-10-19:
Hi, I am having a connection issue, where on connection to my BLE chip it often hangs for sometimes up to 1 min. It would be very useful to have the ability to stop connecting and retry or to timeout the process so it can be restarted ? Any Ideas how to do this ? Thanks !!

dylanrynhart, 2017-10-20:
I read through the ios code and discovered that if you run the disconnect function, it stops current connection attempts, so I was able to achieve this by wrapping this in a timeout. I haven't checked the Android part, but I'm assuming it's similar ?

dylanrynhart, 2017-10-20:
Warning, this doesn't work on Android, must be a different process, I'll check it out.

don, 2017-10-23:
This is probably related to #425

mirkomancin, 2017-12-22:
the problem has been solved? if I try your solution @dylanrynhart I can't resolve my problem.

don, 2018-01-16:
@mirkomancin no resolution yet. Like @dylanrynhart referenced in the Android documentation,  disconnect() [should cancel the connection](https://developer.android.com/reference/android/bluetooth/BluetoothGatt.html#disconnect()). I'm not sure why that's not working.  Issue #425 proposed adding a timeout but that doesn't look like a simple solution. 

Do you know why your device is so slow to connect?  Can you try enabling the [Bluetooth HCI Snoop Log](https://blog.bluetooth.com/debugging-bluetooth-with-an-android-app) and then looking at it in Wireshark to see why the connection is hanging? Are you able to debug the peripheral?

mirkomancin, 2018-01-16:
I check my device configuration and I find the problem. When I try to connect to a device, it restarts his mcu and Android wait for timeout (about 20seconds). If I remove this bug the connection is immediately set on advertising time. 

However, Android and iOs manage the bluetooth connections in a different way and so I did not notice it for that. Maybe, it would be useful to implement iOs logic on Android, that is to try to connect until a connection has been made, rather than waiting for such a long timeout.

don, 2018-06-07:
Long connection times might be problems with service and characteristic discovery. You could try editing the plugin to only discover the services and characteristics that you care about.

Version 1.2.0 has improvements to the disconnect logic.

closed 2018-06-07 by don