don/cordova-plugin-ble-central #484: Phone connected to BLE device, communicating most of the time, but sometimes not

MarioArkwright, 2017-10-26:
Hello,

I made a Ionic script using cordova-plugin-ble-central, using Android on a Smartphone Galaxy S6. I am connecting to a BLE device with 100% of success every time I launch my app. Most of the time, there is no problem to communicate with the device using write (ble.writeWithoutResponse) and read (ble.startNotification) functions. But sometimes, the communication does not work, whereas the connection is successful. 

I usually close the app using the hardware buttons of the Phone. My problem often happens after my Phone has been in sleep mode for more than 20 minutes. Some other times, it happens randomly, after I have killed and relaunched my app several times. When the problem happens, there is no other way to solve it than kill the app and relaunch it, which is most of the time successful. I am now looking for a way to hinder this problem to happen.

Any help would help me not going crazy!
Thank you in advance,

Mario

ddugue, 2017-10-30:
Which version of android are you using?

When you say the communication doesn't work, do you mean you receive an error? does it just hang? never triggering a callback?

Have you tried disconnecting after such a write error to see if it fixes the problem?

MarioArkwright, 2017-10-30:
Thank you for your reply ddugue,

my version of Android is 6.0.1. 

Concerning the communication, when the problem occurs, I do not get any error notification when I am using ble.writeWithoutResponse or ble.startNotification.  And it seems that in both cases neither the success nor the failure functions are fired. 
And whereas the success functions of ble.startScan and then ble.connect are successflly fired, no communication is established between my phone and my BLE device.

When the problem occurs, If I disconnect the BLE via a button calling ble.disconnect function, and then manually disable BLE, enable it again after a few seconds and connect again using a button calling ble.connect, the problem remains. There is no way for me to get rid of this but kill the app and launch it again.

Mario

ddugue, 2017-10-30:
Alright,

I've noticed that on Android, the disconnection doesn't clean up the command stack, so if a command hangs and you disconnect, it won't clean up the command stack.

I've implemented a fix here #481 
It wouldn't solve the issue where it hangs, but at least, when disconnecting, it should start on working again.

For the command hanging, try to run adb logcat to see if any error is raised by the Android firmware, report here and maybe I can help you.

MarioArkwright, 2017-10-31:
I have forked "manually" the 5 files you changed from the plugin, according to issue #481, I hope I made it correctly. Unfortunately, despite this update there is still no way to get rid of my problem but killing my app and launching it again. Disconnecting and connecting it again does not work at this stage.

According to your suggestion, I ran adb logcat and reported the results in the two attached files corresponding respectively to a working and to a not working configs (I removed all the data obviously not concerning the issue as the messages I display in my console, etc.)  . I am sorry that the files contain a lot of data. NB: askInterval is a function I use periodically to send write requests to my device.

I isolate below the lines that seem to illustrate the problem in the config which does not work. This occurs at the first ble.writeWithoutResponse iteration of my app, once the device is successfully connected:

10-31 14:12:38.671 12521 12592 D Peripheral: Write No Response 6e400002-b5a3-f393-e0a9-e50e24dcca9e
10-31 14:12:38.671 12521 12592 E PluginManager: Uncaught exception from plugin
10-31 14:12:38.671 12521 12592 E PluginManager: java.lang.NullPointerException: Attempt to invoke virtual method 'java.util.List android.bluetooth.BluetoothGattService.getCharacteristics()' on a null object reference
10-31 14:12:38.671 12521 12592 E PluginManager:         at com.megster.cordova.ble.central.Peripheral.findWritableCharacteristic(Peripheral.java:569)
10-31 14:12:38.671 12521 12592 E PluginManager:         at com.megster.cordova.ble.central.Peripheral.writeCharacteristic(Peripheral.java:535)
10-31 14:12:38.671 12521 12592 E PluginManager:         at com.megster.cordova.ble.central.Peripheral.processCommands(Peripheral.java:651)
10-31 14:12:38.671 12521 12592 E PluginManager:         at com.megster.cordova.ble.central.Peripheral.queueCommand(Peripheral.java:621)
10-31 14:12:38.671 12521 12592 E PluginManager:         at com.megster.cordova.ble.central.Peripheral.queueWrite(Peripheral.java:592)
10-31 14:12:38.671 12521 12592 E PluginManager:         at com.megster.cordova.ble.central.BLECentralPlugin.write(BLECentralPlugin.java:410)
10-31 14:12:38.671 12521 12592 E PluginManager:         at com.megster.cordova.ble.central.BLECentralPlugin.execute(BLECentralPlugin.java:196)
10-31 14:12:38.671 12521 12592 E PluginManager:         at org.apache.cordova.CordovaPlugin.execute(CordovaPlugin.java:117)
10-31 14:12:38.671 12521 12592 E PluginManager:         at org.apache.cordova.CordovaPlugin.execute(CordovaPlugin.java:98)
10-31 14:12:38.671 12521 12592 E PluginManager:         at org.apache.cordova.PluginManager.exec(PluginManager.java:132)
10-31 14:12:38.671 12521 12592 E PluginManager:         at org.apache.cordova.CordovaBridge.jsExec(CordovaBridge.java:57)
10-31 14:12:38.671 12521 12592 E PluginManager:         at org.apache.cordova.engine.SystemExposedJsApi.exec(SystemExposedJsApi.java:41)
10-31 14:12:38.671 12521 12592 E PluginManager:         at org.chromium.base.SystemMessageHandler.nativeDoRunLoopOnce(Native Method)
10-31 14:12:38.671 12521 12592 E PluginManager:         at org.chromium.base.SystemMessageHandler.handleMessage(SystemMessageHandler.java:39)
10-31 14:12:38.671 12521 12592 E PluginManager:         at android.os.Handler.dispatchMessage(Handler.java:102)
10-31 14:12:38.671 12521 12592 E PluginManager:         at android.os.Looper.loop(Looper.java:158)
10-31 14:12:38.671 12521 12592 E PluginManager:         at android.os.HandlerThread.run(HandlerThread.java:61)

Thank you again for your help,

Mario

[adbLogCatWorkingMin.txt](https://github.com/don/cordova-plugin-ble-central/files/1430894/adbLogCatWorkingMin.txt)
[adbLogCatNotWorkingMin.txt](https://github.com/don/cordova-plugin-ble-central/files/1430893/adbLogCatNotWorkingMin.txt)

ddugue, 2017-10-31:
Thanks for the log,

I've been able to understand what causes the issue. 
It seems you are trying to write to a non-existing service and that makes the plugin crash. I've submitted a PR #486 , but even if you install it, it wouldn't make a difference for your case (except that it wouldn't crash, it would simply return an error in your callback)

You need to make sure that the service you are writing to exists and is available on the BLE

MarioArkwright, 2017-10-31:
Thank you ddugue for this reply, 

I fetched your modifications from #486, but I am afraid I still receive no error message when the problem occurs, and the plugin still crashes. 

I also wrote this in my code:

ble.connect(
          device.id,
          function (connectSuccess) {                                   
            console.log(JSON.stringify(connectSuccess, null, 2))
            [...]
          }
)

and what I can read in my console is the same things, whether the bug appears or not:

{
  "name": "blabla",
  "id": "DF:C1:53:43:21:AC",
  "advertising": {},
  "rssi": -84,
  "services": [
    "1800",
    "1801",
    "6e400001-b5a3-f393-e0a9-e50e24dcca9e"
  ],
  "characteristics": [
    {
      "service": "1800",
      "characteristic": "2a00",
      "properties": [
        "Read",
        "Write"
      ]
    },
    {
      "service": "1800",
      "characteristic": "2a01",
      "properties": [
        "Read"
      ]
    },
    {
      "service": "1800",
      "characteristic": "2a04",
      "properties": [
        "Read"
      ]
    },
    {
      "service": "1800",
      "characteristic": "2aa6",
      "properties": [
        "Read"
      ]
    },
    {
      "service": "1801",
      "characteristic": "2a05",
      "properties": [
        "Indicate"
      ],
      "descriptors": [
        {
          "uuid": "2902"
        }
      ]
    },
    {
      "service": "6e400001-b5a3-f393-e0a9-e50e24dcca9e",
      "characteristic": "6e400002-b5a3-f393-e0a9-e50e24dcca9e",
      "properties": [
        "WriteWithoutResponse",
        "Write"
      ]
    },
    {
      "service": "6e400001-b5a3-f393-e0a9-e50e24dcca9e",
      "characteristic": "6e400003-b5a3-f393-e0a9-e50e24dcca9e",
      "properties": [
        "Notify"
      ],
      "descriptors": [
        {
          "uuid": "2902"
        }
      ]
    }
  ]
}

As the service and characteristic seem to be found (and indeed correspond to the services and characteristics I intend to use, i.e. "6e400001-b5a3-f393-e0a9-e50e24dcca9e" etc.), I do not understand why the plugin fails to write and read.

I hope these info help. 

Best,

Mario

ddugue, 2017-10-31:
Humm, intriguing.

Can you try running your app with: https://github.com/ddugue/cordova-plugin-ble-central/tree/mario

and send the adb logcat logs?
Also could you share with me the snippet of where you are calling write no response in your js?
Thanks,

MarioArkwright, 2017-11-02:
Hello ddugue,

please find attached a .zip file containing the adb logcat logs using the version of the plugin you just sent me, and the controllers.js and services.js files that I wrote to communicate via ble (I am using AngularJS and Cordova via Ionic). The first call to ble.writeWithoutResponse appears in line 183 of controllers.js, as a result of the connectSuccess function of ble.connect.

Thank you,

Mario 
  
[debug2.zip](https://github.com/don/cordova-plugin-ble-central/files/1437156/debug2.zip)

ddugue, 2017-11-08:
First off I've noticed that the new adb logcat are not using the version I sent you.

You can see it by looking at line 535 of the logs. It writes "Write No Response xxx" instead of the" Write No Response Char xxx" ([https://github.com/ddugue/cordova-plugin-ble-central/blob/mario/src/android/Peripheral.java#L672](https://github.com/ddugue/cordova-plugin-ble-central/blob/mario/src/android/Peripheral.java#L672))

But at least, I think I've found the source of your issue.

I've read that the UART service you are using simply doesn't support writeWithoutResponse (it is a different Bluetooth call than write). You could easily test this by replacing the ble.writeWithoutResponse with ble.write (and providing a callback, even if just writes to console).

Hope it helps!

MarioArkwright, 2017-11-10:
Hello ddugue,

according to your last reply, I replaced all my ble.writeWrithoutResponse(...) with ble.write(...) in my code. I am afraid the problem still happens alas. I also analyzed my BLE peripheral with the commercial ap BLE Scanner, and according to it - see screenshot in attached file - the ble.writeWrithoutResponse(...) should be supported.

In parallel, I developped another app with exactly the same code architecture, to communicate with [this Arduino compatible board](https://www.adafruit.com/product/3406) including [the same Nordic SoC](https://www.nordicsemi.com/eng/Products/Bluetooth-low-energy/nRF52832) as in my former device, with which the bug appears. Using this config, my bug does never happen. It is then possible that something turned wrong with the SDK (provided by Nordic) or the way I am using it. What is all the more weird though, is that with a few Phones the bug does not happen. 

If I can get rid of the bug after reconsideration of the Nordic SDK, I will let you know.
In any case, again I thank you a lot for your helpful analysis.

Best regards,

Mario 

[Screenshot.zip](https://github.com/don/cordova-plugin-ble-central/files/1461118/Screenshot.zip)

ddugue, 2017-11-13:
Hey @MarioArkwright ,

Sorry I could not help you more. Meanwhile, if you do find an issue with the plugin, it would be nice of you to report it, so we can fix it.

Just so you know, you could try modifying your local version of the plugin (to do tests or report adb logcat values with `log`). I don't know if it works the same in Ionic (it should since it wraps around cordova), but in cordova, you need to modify the code of the plugin in the node_modules folder and sometimes in the plugins folder. Then you have to remove the android platform and re-add it, else it won't work.

I really hope you can fix your issue.
Cheers,

don, 2018-01-16:
Based on the stack trace, I'd guess the service UUID is being passed into ble.writeWithoutResponse is null or doesn't exist on that peripheral.

    10-31 14:12:38.671 12521 12592 D Peripheral: Write No Response 6e400002-b5a3-f393-e0a9-e50e24dcca9e
    10-31 14:12:38.671 12521 12592 E PluginManager: Uncaught exception from plugin
    10-31 14:12:38.671 12521 12592 E PluginManager: java.lang.NullPointerException: Attempt to invoke virtual method 'java.util.List android.bluetooth.BluetoothGattService.getCharacteristics()' on a null object reference
    10-31 14:12:38.671 12521 12592 E PluginManager: at com.megster.cordova.ble.central.Peripheral.findWritableCharacteristic(Peripheral.java:569)
    10-31 14:12:38.671 12521 12592 E PluginManager: at com.megster.cordova.ble.central.Peripheral.writeCharacteristic(Peripheral.java:535)

The service is null here https://github.com/don/cordova-plugin-ble-central/blob/155827ff192d0f87947971c7cbf6f23f0e46de5b/src/android/Peripheral.java#L569

The service is looked up from the GATT by UUID https://github.com/don/cordova-plugin-ble-central/blob/155827ff192d0f87947971c7cbf6f23f0e46de5b/src/android/Peripheral.java#L534-L535

Which is passed into writeCharacteristic https://github.com/don/cordova-plugin-ble-central/blob/155827ff192d0f87947971c7cbf6f23f0e46de5b/src/android/Peripheral.java#L525

I should have better null checks, but I don't so it's up to you to ensure you're passing in a valid data.  You could open your Cordova project ($PROJECT/platforms/android) with Android Studio and set a breakpoint in the debugger confirm this.

closed 2018-06-07 by don