don/cordova-plugin-ble-central #485: Intercept Advertising during app is in foreground or in background

mburger81, 2017-10-31:
Hi, this is not an issue but this is a feature request or a question

We have a HealthCare device which is measuring Heart Rate and others, the device is advertising with a custom serviceUUID and a custom characteristicUUID the data.

At the moment we must scan for new devices, connect to the device send command an wait for advertising from device to the the information.

Is there a way to listen to bluetooth devices come UP and visible to device and do a connecting there?

startScanWithOptions is scanning always which could be a good solution, but it is reposring the device only for the first time, so if we shutdown the device and wake it up the event is never fired.
If we use reportDuplicates paramter until the device is on we have many many events for same device, not only on off/on but the whole time the device is on there are at least 20 and more events!

Can we do something like that?

ddugue, 2017-10-31:
The problem with scanning is it is a broadcast, so let's say you scan for devices with reportDuplicates, you monitor the events and notice that one of the devices stop appearing. It could mean one of three things:

1. The device is off
2. The device is out of range
3. The device is connected and is not broadcasting when connected to.

There is no real way of knowing if the device is out of range / not broadcasting OR if the device is off with your phone. If you have access to the dongle firmware, you could implement a flag that is visible in the scanning data that show if the device has ever been connected to since it powered on. You could then connect to it when you detect that flag in the scan callback.

The other solution would be to listen to the scan callback with report duplicates and when a device doesn't appear for a while you remove it from your internal list in your js app and when it reappears you trigger a custom event specific to your js app.

Hope it answers your question,

mburger81, 2017-11-02:
Hi, sorry for my late answer and thx for your response, today I created a PR on ionic native to support startStateNotifications and stopStateNotifications :) 
https://github.com/ionic-team/ionic-native/pull/2085

I'll have a look to your suggestions and have a try to implement something like that.

If I see how the app from the device producer is working, it is doing the thing I wan't to implement.
I open the app, I power on the device and anything magical is working in background, the app is recognizing the device, connect to it and download the health care measurement.

I have to questions I would like to ask you:
a) if I do a startScanWithOptions passing the serviceUUID which is the the customized service from device I use also to get notifications, the startScanWithOptions is never firing an event.
`.startScanWithOptions(['00001523-1212-efde-1523-785feabcd123'], { reportDuplicates: true })`
Why could that be?
b) what does the `startScan/WithOptions` and `stopScan` do under the hoot? If we have enabled Bluetooth on android device, the device is always scanning for devices, for example if I have my smart watch it is always discovered and connected in the background out of the box without need of do it manually. So the question is, is startScan a software layer from you which is only notifying seeing the a device? Or does it something which does consume more battery then the default behavior for smart watches or speaker scanning?

Can you help? Thx!