don/cordova-plugin-ble-central #487: Android: Command queue not flushed on disconnect

lcrocker, 2017-11-08:
Read/write commands are queued on Android; "success" of the commands is reported when they are queued, not when they are actually performed; and disconnecting does not flush the queue. So it's possible (and I have seen) to issue a write command, get the success callback, then disconnect without the write actually having been performed.

I work around this by simply delaying my disconnect a bit, but it should perhaps be rethought. Either flush the queue on disconnect, or don't report success until the write is actually performed (IMHO the latter would be better, though perhaps harder to implement).

ddugue, 2017-11-09:
Hey @lcrocker , I implementend the fix in #481 , just waiting for it to be merged upstream!

don, 2018-01-16:
@lcrocker on Android the success commands are called when the [BluetoothGattCallback](https://developer.android.com/reference/android/bluetooth/BluetoothGattCallback.html) callbacks are called after the command completes, not when it's queued.

for read https://github.com/don/cordova-plugin-ble-central/blob/155827ff192d0f87947971c7cbf6f23f0e46de5b/src/android/Peripheral.java#L259
for write https://github.com/don/cordova-plugin-ble-central/blob/155827ff192d0f87947971c7cbf6f23f0e46de5b/src/android/Peripheral.java#L279

Are you seeing this with write or writeWithoutResponse? It's possible Android cheats with writeWithoutResponse and returns success right away? If you have an app that can duplicate this, it would help track the issue down.

don, 2018-01-16:
@ddugue I still need to review your pull request. Sorry.

don, 2018-06-07:
Fixed as part of #562

closed 2018-06-07 by don