don/cordova-plugin-ble-central #488: 1128 Bluetooth UHF RFID Reader + Cordova plugin

Rubathas, 2017-11-09:
Hi,

I am trying to investigate whether I should go native only (iOS) or Cordova (Ionic) for an app we are trying to develop for the [1128 Bluetooth UHF RFID Reader](www.tsl.uk.com/products/1128-bluetooth-handheld-uhf-rfid-reader/).

The manufacturer has only provided native SDKs (e.g. for iOS, Android) and one SDK for Xamarin. There are no Cordova/JS libraries.

Hence, my question is - would it be possible to use this plugin together with the native iOS SDK and develop a hybrid app communicating with this device and send the ASCII commands to update RFID tags?

Or is your recommendation to go purely native with iOS? I see that other people have tried to go down this road with a different Cordova plugin, but have been directed to this plugin due to the device being "Bluetooth Classic" (e.g. https://github.com/don/BluetoothSerial/issues/148).

I would appreciate any feedback - thanks in advance.

Sincerely,
Rubathas

ddugue, 2017-11-09:
I'm not a maintainer of this plugin, but maybe I can answer from my personal experience.

It will not be easy to merge a native SDK with this plugin. I think several collaborators are trying to bring a kind of sharing mechanism with this plugin to implement native SDK more seamlessly. It is far from being completed though

It also boils down to your programming experience and wether you want to target multiple platforms. Theoritically speaking, you could probably implement your native SDK functionnalities in Javascript and use this plugin. This plugin implements a proxy to (almost) all the native calls of either iOS and android. But if you have more iOS experience and you are only targeting iOS, then I guess you should go native.

Hope it helps you in your decision.

don, 2018-01-16:
@Rubathas since the device is Bluetooth Classic you can't use this plugin. MFi is a pain, but the good news is that is the only way to make Bluetooth Classic work on iOS.

You could still use Ionic if you want, but you'd need to write a new plugin. Since TSL provides native libraries, you can design a JavaScript API then implement the plugin using the native libraries. I done this for many devices and it usually works well.

You'll still need to go through the approval process for MFi on iOS, but it won't matter than you're using Ionic or Cordova.

closed 2018-01-16 by don