don/cordova-plugin-ble-central #489: Bluetooth device disconnects after 10 seconds

evilword666, 2017-11-09:
Hi, I´m trying to connect a BLE localizador, I have an Ionic app, in this one I use your BLE plugin, with this I´m able to connect all the device services y features.  Nevertheless, this connection stablished with a certain time, after 10 seconds the connection is loss. I have an Android implement app; this app allows a constant connection with the device and obtain without problems: services and features.  I have checked out the Android application and this use a library in this one the application creates a service. The service manages the Bluetooth connections and allows the constant connection between the app and the device. Also, this service uses the class IBinder. My question is: Do you know some way to allow the Bluetooth connection with the localizator and  avoid the loss of connection using IBinder in Ionic?

anjildhamala, 2017-11-10:
unless your peripheral kicks out the central, it should not lose connection. I've never had this issue with this plugin.

don, 2018-01-16:
@evilword666 Can you see if apps like [LightBlue Explorer](https://itunes.apple.com/us/app/lightblue-explorer/id557428110?mt=8) or [nRF Connect](https://itunes.apple.com/us/app/nrf-connect/id1054362403?mt=8) also disconnect after 10 seconds?

Are you seeing any error message in XCode or Safari web developer tools when device disconnects?

closed 2018-06-07 by don