don/cordova-plugin-ble-central #491: how to print Chinese?

wjm117, 2017-11-11:
Hi ,this plugin is very useful for me ,but there is something wrong.When I print English or numbers, everything is normal, but garbled when printing Chinese。So, how can i do for use plugin print Chinese? thinks

ddugue, 2017-11-13:
Hey @wjm117 ,
When you say printing Chinese character, you mean Chinese character in the dongle name?

don, 2018-01-16:
@wjm117 If you're sending Chinese characters you need to make sure you encode them correctly before sending not just sending ascii bytes like English. Check out [bytesToString](https://github.com/don/phonegap-nfc/blob/master/www/phonegap-nfc.js#L520-L565) and [stringToBytes](https://github.com/don/phonegap-nfc/blob/master/www/phonegap-nfc.js#L567-L596). Note that those examples are old and use arrays of ints instead of ArrayBuffers.

closed 2018-05-07 by don

B200019A, 2023-04-14:
HI, i using your method(stringToBytes) but the thermal printer print the number only, what are the problem?

 this is my code
 var test = "你好\n";

  var encodeText = new TextEncoder("GB18030", {
        NONSTANDARD_allowLegacyEncoding: true,
    }).encode(stringToBytes(test)+"\r\n\u001B\u0070\u0000\u000F\r\n");

  ble.write(
    printer_id,
    "49535343-FE7D-4AE5-8FA9-9FAFD205E455",
    "49535343-8841-43F4-A8D4-ECBE34729BB3",
    encodeText.buffer,
    (success) => {
      console.log(success + ": print" + test);
    },
    (failure) => {
      console.log(failure);
    }
  );
  
  const stringToBytes =(string) => {
     // based on http://ciaranj.blogspot.fr/2007/11/utf8-characters-encoding-in-javascript.html

     var bytes = [];

     for (var n = 0; n < string.length; n++) {

         var c = string.charCodeAt(n);

         if (c < 128) {

             bytes[bytes.length]= c;

         } else if((c > 127) && (c < 2048)) {

             bytes[bytes.length] = (c >> 6) | 192;
             bytes[bytes.length] = (c & 63) | 128;

         } else {

             bytes[bytes.length] = (c >> 12) | 224;
             bytes[bytes.length] = ((c >> 6) & 63) | 128;
             bytes[bytes.length] = (c & 63) | 128;

         }

     }

     return bytes;
 }