don/cordova-plugin-ble-central #492: Callback not run on disconnect

alanstallwood, 2017-11-13:
Hi,

I'm running android 8 on a Pixel C tablet. When bluetooth is turned off the disconnect callback method is not run.

No such issue exists on Pixel XL phone running android 8 or other android/ios devices.

ddugue, 2017-11-13:
From what I can gather, it seems to be an issue with a bluetooth profile set up on the phone that listen to the intents and try to unbind an unexisting connection from the Context.

Follow up questions:
- Is there any other bluetooth app on the Pixel C?
- Does the dongle you are trying to connect support the Bluetooth MAP profile?
- What operations are you doing on the dongle before disconnecting?
- Do you have any bluetooth paired device to your pixel C (Car system, bluetooth audio, ...)

In any case, I do not think it is related to this plugin, but I would like to get to the bottom of this issue to learn more about all the weird Android behaviors. Especially if another bluetooth app can interfere with another bluetooth app functionality.

alanstallwood, 2017-11-13:
Nothing else is using bluetooth
I'm not sure about MAP. We're using GATT to communicate.
The device periodically sends messages to the tablet every 30 seconds or less. Each message is acknowledged by the tablet. I'm turning bluetooth off on the tablet to simulate a disconnect.
Nothing else paired

ddugue, 2017-11-13:
Since you are turning bluetooth off to simulate disconnection, how are you providing the disconnect callback?

When are you turning the bluetooth off? Before calling ble.disconnect or after calling ble.disconnect?

alanstallwood, 2017-11-14:
The callback function is passed as the 3rd parameter to ble.Connect

ble.Disconnect isn't called at all.

ddugue, 2017-11-14:
Oh, so it is in fact the connect callback error which is not called!

Instead of turning the bluetooth off on the tablet to simulate a bad connection, can you try turning off the bluetooth device?

ddugue, 2017-11-14:
Alright,
So, _normally_ android should fire the disconnect event, but that implementation is firmware specific. It is entirely possible that the tablet firmware, upon disabling bluetooth, clear the whole memory linked to the bluetooth stack. Including the callbacks.

A solution for your app, would be to listen to the state notifications with `ble.startStateNotifications` and `ble.stopStateNotifications` and trigger your callback code when a connection is active and the bluetooth gets shutted down.

Meanwhile, that feature (triggering connection callback error, when disabling bluetooth) could be implemented in the plugin quite easily. I don't mind doing it, we just need to make sure it wouldn't cause any backward compatibility issue.

@don what do you think?

don, 2018-01-16:
@ddugue using state notifications is a good workaround. I'm fine if you patch the plugin. There are some other disconnect problems with some devices. See #412 and others.

closed 2018-06-07 by don

Jatindersyall12, 2023-08-16:
@don 

I am facing connection callback issue. I want to connect two BLE devices with GATT server but below code say's it's already connected, but my devices never connected. Can you please help me out on this.

bluetoothGatt = device.connectGatt(
            this,
            true,
            object : BluetoothGattCallback() {
                override fun onConnectionStateChange(gatt: BluetoothGatt, status: Int, newState: Int) {
            
                    when (newState) {
                        BluetoothProfile.STATE_CONNECTED -> {
                            gatt.discoverServices()
                        }
                        BluetoothProfile.STATE_DISCONNECTED -> {

                            //deviceStatusListener?.onDisconnect()
                            //closeConnection()
                        }
                    }

Looking for your's valuable feedback soon.

Thanks Jatinder Syall

peitschie, 2023-08-16:
Hi @Jatindersyall12,

Would you mind raising a new issue to discuss this?

Thanks!

Jatindersyall12, 2023-08-16:
@peitschie created a new issue for this.