don/cordova-plugin-ble-central #493: ble.writeWithoutResponse not working on ios 11.0.1 please help

aaviles24, 2017-11-16:
My code works fine on Android but then when I try it on ios 11.0.1 it does not work.
I hard coded UUID just to simplify my post but I do use ble.startScan to retrieve UUID and I am able to connect and start notification on another characteristic just fine. It is only when I go to writeWithoutResponse that it shows a successful write but nothing is received by the device.

I have also verified using apps Dialog DSPS and Blue Gecko to verify that writeWithoutResponse works with the UUID's below.

I have included below my code and console outputs in xcode showing that the serialPortCharacter2UUID does have the writeWithoutResponse property.

I am not sure what else to try at this point is this a known issue. Please help.

var devUUID = "B6EDADC8-4244-0753-766B-08CFB6EEA058";
var serviceUUID   = "0783B03E-8535-B5A0-7140-A304D2495CB7";
var serialPortCharacter2UUID = "0783B03E-8535-B5A0-7140-A304D2495CBA";

var timeResponse = "1510870330\r";
var array = new Uint8Array(timeResponse.length);
for (var i = 0, l = timeResponse.length; i < l; i++) {
    array[i] = timeResponse.charCodeAt(i);
}
var finalArr = window.btoa(String.fromCharCode.apply(null, array));

var success = function(){
 console.log("Successfully executed writeWithoutResponse");
}

var failure= function(){
 console.log("Failed to execute writeWithoutResponse");
}

ble.writeWithoutResponse(app.device,serviceUUID, serialPortCharacter2UUID, finalArr, success, failure);

2017-11-16 16:33:08.711257-0500 Scarab[1029:889444] Discovered {
    advertising =     {
        kCBAdvDataIsConnectable = 1;
        kCBAdvDataLocalName = SCARAB;
        kCBAdvDataServiceUUIDs =         (
            "0783B03E-8535-B5A0-7140-A304D2495CB7"
        );
    };
    id = "B6EDADC8-4244-0753-766B-08CFB6EEA058";
    name = SCARAB;
    rssi = "-52";
}

2017-11-16 16:37:29.423090-0500 Scarab[1029:889444] connect
2017-11-16 16:37:29.423105-0500 Scarab[1029:889444] Connecting to peripheral with UUID : B6EDADC8-4244-0753-766B-08CFB6EEA058
2017-11-16 16:37:29.587105-0500 Scarab[1029:889444] didConnectPeripheral
2017-11-16 16:37:29.945629-0500 Scarab[1029:889444] didDiscoverServices
2017-11-16 16:37:30.188931-0500 Scarab[1029:889444] didDiscoverCharacteristicsForService
2017-11-16 16:37:30.189490-0500 Scarab[1029:889444] Found characteristics for service <CBService: 0x1c0468780, isPrimary = YES, UUID = 0783B03E-8535-B5A0-7140-A304D2495CB7>
2017-11-16 16:37:30.189646-0500 Scarab[1029:889444] Characteristic <CBCharacteristic: 0x1c40bb180, UUID = 0783B03E-8535-B5A0-7140-A304D2495CB8, properties = 0x10, value = (null), notifying = NO>
2017-11-16 16:37:30.189883-0500 Scarab[1029:889444] Characteristic <CBCharacteristic: 0x1c40b9f20, UUID = 0783B03E-8535-B5A0-7140-A304D2495CBA, properties = 0x4, value = (null), notifying = NO>
2017-11-16 16:37:30.190144-0500 Scarab[1029:889444] Characteristic <CBCharacteristic: 0x1c40ba040, UUID = 0783B03E-8535-B5A0-7140-A304D2495CB9, properties = 0x16, value = (null), notifying = NO>
2017-11-16 16:37:30.291490-0500 Scarab[1029:889444] connect success
2017-11-16 16:37:30.291758-0500 Scarab[1029:889444] registering for notification
2017-11-16 16:37:30.291793-0500 Scarab[1029:889444] getData
2017-11-16 16:37:30.292072-0500 Scarab[1029:889444] Looking for 0783B03E-8535-B5A0-7140-A304D2495CB8 with properties 16
2017-11-16 16:37:30.335475-0500 Scarab[1029:889444] didUpdateValueForCharacteristic
2017-11-16 16:37:30.338282-0500 Scarab[1029:889444] didUpdateValueForCharacteristic
2017-11-16 16:37:30.364831-0500 Scarab[1029:889444] didUpdateValueForCharacteristic
2017-11-16 16:37:30.395817-0500 Scarab[1029:889444] didUpdateValueForCharacteristic
2017-11-16 16:37:30.455558-0500 Scarab[1029:889444] didUpdateValueForCharacteristic
2017-11-16 16:37:30.545812-0500 Scarab[1029:889444] didUpdateValueForCharacteristic
2017-11-16 16:37:30.579644-0500 Scarab[1029:889444] writeWithoutResponse
2017-11-16 16:37:30.579662-0500 Scarab[1029:889444] getData
2017-11-16 16:37:30.579814-0500 Scarab[1029:889444] Looking for 0783B03E-8535-B5A0-7140-A304D2495CBA with properties 4
2017-11-16 16:37:35.080554-0500 Scarab[1029:889444] didUpdateValueForCharacteristic
2017-11-16 16:37:35.106900-0500 Scarab[1029:889444] didUpdateValueForCharacteristic
2017-11-16 16:37:35.157934-0500 Scarab[1029:889444] didUpdateValueForCharacteristic
2017-11-16 16:37:35.167258-0500 Scarab[1029:889444] didUpdateValueForCharacteristic
2017-11-16 16:37:35.226691-0500 Scarab[1029:889444] didUpdateValueForCharacteristic
2017-11-16 16:37:35.256631-0500 Scarab[1029:889444] didUpdateValueForCharacteristic
2017-11-16 16:37:35.287302-0500 Scarab[1029:889444] didUpdateValueForCharacteristic
2017-11-16 16:37:35.317554-0500 Scarab[1029:889444] didUpdateValueForCharacteristic
2017-11-16 16:37:35.376824-0500 Scarab[1029:889444] didUpdateValueForCharacteristic

ddugue, 2017-11-20:
Hey @aaviles24 ,

So your success callback is never called right, and ONLY on ios? Do your characteristic support write (with response)? Did you try with write instead?

What other operations are you doing before writing that data? Are you registering for notification, reading or writing other data?

aaviles24, 2017-11-20:
Success Callback is called. In both IOS and Android. No it does not support write with response. I did try with just write instead but no success and again write with response is not supported.

Before writing that data I subscribe to notifications of another characteristic on that same service UUID and I am receiving notifications for the other Characteristic. I have tried forgoing the subscription to notifications to see if that was interfering and no luck.

I have also verified that writewithoutResponse works on this specific characteristic using Blue Gecko ios app and Dialog DSPS ios app.

aaviles24, 2017-11-20:
I am also looking through the plugin code but I cant find anything that would cause this symptom.

don, 2018-01-16:
@aaviles24 your code looks ok and Looking for 0783B03E-8535-B5A0-7140-A304D2495CBA with properties 4 shows it supports [write without response](https://developer.android.com/reference/android/bluetooth/BluetoothGattCharacteristic.html#PROPERTY_WRITE_NO_RESPONSE). 

I'm not sure what's wrong? Is this working with newer versions of iOS 11.x? Do you see the problem on every iOS device?

kim0z, 2018-04-18:
I have the same issue, any luck @aaviles24 ?  
2018-04-18 20:39:26.014027+0300 GreatBear[6961:1945905] writeWithoutResponse
2018-04-18 20:39:26.014074+0300 GreatBear[6961:1945905] getData
2018-04-18 20:39:26.014423+0300 GreatBear[6961:1945905] Looking for 00001532-1212-EFDE-1523-785FEABCD123 with properties 4

chicag0, 2018-04-26:
@don 

I'm having issues with ble.write on iOS 11.

On android the success callback works great, but on iOS neither the success nor failure callback are firing.  

`ble.write(bluetooth_id, service_uuid, characteristic_uuid, data,
    function(response) {
      alert(JSON.stringify(response, null, 2));

    }, function(response) {
      $('#ble_response').val(response);
       alert(JSON.stringify(response, null, 2));
    }
  );`

don, 2018-04-26:
@kim0z and @chicag0 make sure you're using the correct function write vs writeWithoutResponse.  The peripheral determines the write type. Check the details in the JSON returned when you connect. Some versions of Android were more flexible about the write type. iOS requires the correct type. Also, iOS agressively caches peripheral information. If the characteritics have changed, make sure you cycle Bluetooth on and off so you get the latest information.

chicag0, 2018-04-26:
@don 

At least on my test app, both ble.writeWithoutResponse and ble.write return "OK" on Android, but iOS doesnt seem to do anything.  Digging in deeper now but thought I would chime in here.

kim0z, 2018-04-29:
in my case I was using wrong character, now it works.

don, 2018-05-07:
@kim0z thanks for the update
@chicag0 some android versions appear to do the right thing no matter which write function is called. In my experience iOS requires you to call the correct function based on the characteristic properties.

closed 2018-05-07 by don