don/cordova-plugin-ble-central #494: Scan not working in crowded places

MattiaCostamagna, 2017-11-20:
Hi Don,

We are developing an application using Ionic2 that will manage mixers via bluetooth.
It's working fine, it lists the devices that need to be listed, it connects to them and sends the commands correctly.

However, we've had some problems when using our app in places where there are a lot of bluetooth devices: in this case the startScan method does not find any suitable device before calling the stopScan after 20 seconds, so I cannot connect to the device I want and I cannot use my app.

I understand that with more devices it takes more time to list and display them, but if we use an external (BLE Scanner or BLE scan raw) it displays all the devices in a couple of seconds.

I've tested this app with Android (Moto X+1, OS version 6.0) and iOS (OS version 11.1.1) and it gives the same result (works fine only when there are few bluetooth devices).

Is there a solution to this?
Thank you in advance.

Mattia

ddugue, 2017-11-20:
If you are willing to get your hand dirty a bit, you could try implementing the scan with https://developer.android.com/reference/android/bluetooth/le/BluetoothLeScanner.html

It allows finer grain control over the scan options. I do not mind doing it, but I want @don green light before, as it would take some time to implement

MattiaCostamagna, 2017-11-23:
I've got some news: 
In my application I was previously pushing the devices that my ble.scan found only if their names' length was greater than 0 and started with a string we defined. This gave me a lot of problems since I didn't even get the device I was looking for. 

I've just removed the two ifs statement described above, and let my app lististing every device it finds.
It works like a charm, and I have no clue of why that two ifs were giving me so much trouble.
I have now moved the statement from the ts class to the angular2 template, in order to display only what I want, and it's working (at least better than before).

I realy don't know how this is possible, and I honestly thisk that there is something else, but the tests I've done prove otherwise.

Anyhow I'm curious of what could be the reason of how it is working now, so if anyone has got a clue, please let me know.

MattiaCostamagna, 2017-11-23:
Another thing: now it is working nice and it finds always the devices. But if, for any reason, the app does not find a device the first time, it won't find it immediately, but after many attempts, as if it had an internal cache.

I'm resetting the array in which I'm pushing the devices I'm discovering every time I launch a new scan, so it starts clean.

This is the javascript method I'm calling when refreshing the page:
```javascript
let self = this;
self.ble
    .startScan([])
    .subscribe((value) => {
        //if (value.name.length > 0) {
        self.asyncDevices.push(value);
        //}
    });

setTimeout(function () {
    self.ble
        .stopScan()
        .then(() => {
            self.refreshing = false;
            self.starting = false;
            if (refresher) {
                refresher.complete();
            }
            /*
            ...
            other unimportant stuff
            ...
            */
        });
}, BluetoothData.ScanTime);
```

And this is the Angular2 template page with the ionic list to which I'm posting the found devices:
```html
<ion-list>
    <div *ngFor="let device of asyncDevices">
        <button class="device" ion-item *ngIf="device.name && device.name.length>0" (click)="selectDevice(device)">
                {{ device.name }}
            </button>
    </div>
</ion-list>
```

don, 2018-01-16:
@MattiaCostamagna thanks for the update. Some devices don't advertise very often so you might miss them in crowded areas.

It could be Angular too. If you update the DOM from Bluetooth, Angular doesn't know about it since it occurs outside the Angular lifecycle. Typically I need to wrap these call in NgZone to ensure the UI updates.

Try something like

    this.ngZone.run(() => {
        self.asyncDevices.push(value);
    });

closed 2018-01-16 by don