don/cordova-plugin-ble-central #498: Ionic 3.9.2 BLE not working on IOS

jpiwek, 2017-11-22:
Hi Don,

I am using Ionic 3.9.2 with Pro backend and new Ionic View app (blue icon) on IOS and Android.
BLE works fine in new Ionic View app on Android and not on IOS. 
Before I have switched to Ionic Pro backend and new IONIC View, BLE worked fine on both....
Any hints or experiences you've made?

Best,
Piwi

carlst, 2017-11-27:
Not sure if this is related, but we needed to comment out the JS `enable `objects
in `plugins/cordova-plugin-ble-central/www/ble.js` to avoid errors shown in the Xcode
runtime console log.

-Carl

don, 2018-01-16:
@jpiwek I haven't used Ionic Pro. Since it was working before I'd guess there's problem with the new 
View app, unfortunately you can't run the Safari web tools on that.

Are you still seeing this with the latest versions of that app?

jpiwek, 2018-01-16:
No, with the new version everything is working. Thanks for you help!

closed 2018-01-16 by jpiwek