don/cordova-plugin-ble-central #501: when i'm use the 'ble.write' or 'ble.read' .how do I get the 'service_uuid' and 'characteristic_uuid';

Shaooooo, 2017-11-27:
i need to read the information from other bluetooth equipment. how do I get the params  'service_uuid' and 'characteristic_uuid'

paredesivan, 2017-12-14:
tenes que escanear, conectarte, ver cual es el servicio, caracteristica y despues leer desde ese servicio y caracteristica el valor

don, 2018-01-16:
@Shaooooo the service and characteristic info comes from the Bluetooth device. If you're lucky, you'll find documentation. Alternately, like @paredesivan says you need to scan and discover the services and characteristics.

I use (LightBlue Explorer)[https://itunes.apple.com/us/app/lightblue-explorer/id557428110?mt=8] on iOS and (nRF Connect)[https://play.google.com/store/apps/details?id=no.nordicsemi.android.mcp&hl=en] on Android.

closed 2018-01-16 by don