don/cordova-plugin-ble-central #503: startScanWithOptions is not working in background after some time

SatheeshKumarBalu, 2017-12-06:
**Device**: Android 7.0 / redmi Note 4
**Application**: Running in background
**Permission**: Bluetooth, Location service and Internet are enabled
**Issue**: In background I just stopping the scan and again starting scan after 30sec by using **startScanWithOptions**. but the scan was not started and it was not detecting any devices. In foreground its working fine.
**Reproduce**:  The same issue is happening in some other mobile also. but the issue is not consistent.

don, 2018-01-16:
@SatheeshKumarBalu so this is working on most devices but failing on a few? Are you seeing any errors in `adb logcat` or errors in the JavaScript console with chrome://inspect.

SatheeshKumarBalu, 2018-01-17:
Thanks for your response @don . 
After few more testing we can consistently reproduce the issue in one scenario.

Application running in background -> screen locked -> keep the mobile ideal for 30min -> App logic just try to stop the scan and again start the scan after 30sec by using startScanWithOptions. after that the scan was not started and it was not detecting any devices.

The problem is I can't able to get any error so far, because the test case is like mobile in ideal condition. 
Generally Its working fine in background but **not working in mobile ideal state for 30min**.

don, 2018-06-07:
I don't have an answer for this but we can talk about a solution if you track down the cause. Using the new  [ble.autoConnect()](https://github.com/don/cordova-plugin-ble-central#autoconnect) function be an alternate solution if you're scanning to connect to a device.

closed 2018-06-07 by don