don/cordova-plugin-ble-central #506: ble.connect not working on Android 7.1.1

mardugas, 2017-12-14:
I'm writing a small app to read measurements from a blood pressure device
(service 1810, characteristic 2A35).
It works fine on Android 4.4.
When installed on an Android 7.1.1. device, ble.startScan works ok (ACCESS_COARSE_LOCATION permission is granted & accepted by the user). But ble.connect doesn't work.
Any ideas? Thanks, Martin

FLATT-Dev, 2017-12-22:
I have 2 identical Galaxy 3 phones. One has the latest kernel build, Oct 24 2017, and another is  Oct 16 2016. Connect works on the older phone but does not on the newer one.. I get the error "Peripheral disconnected" from the plugin. I have access to  BLE device firmware and I do not see it ever getting a GATT MTU exchange event or GATT connect event. I'm looking into it, will poist if I find anything

zerox1212, 2018-01-09:
Anyone find out what the problem is? I was trying to avoid writing native apps for my simple BLE theromometer, but `onConnect: function(peripheral)` just returns `null`. `onError: function(reason)` is never called.

paredesivan, 2018-01-12:
@FLATT-Dev in a samsung galaxy s3? that cell phone does not have version 4.3?

paredesivan, 2018-01-12:
estuve mirando la documentacion de android y no menciona cambios en las versiones 7.0. el problema debe venir por otro lado

FLATT-Dev, 2018-01-12:
@paredesivan, the Android version I have on this phone is 6.0.1. I  dug a little deeper found that  onConnectionStateChange callback returns status 133. Some posts online suggest putting the connection attempt  on a UI thread. Tried that and it did not help. Strangely, I have an identical phone where connection does work. So I'm befuddled

zerox1212, 2018-01-12:
Eventually my BLE on Android stopped returning null to `onConnect` and started working. It seems like when I first installed PhoneGap Developer App and ran my app for the first time (I selected "Allow" on location) `onConnect` would return null. After I closed the app entirely and ran it again it is now working. For me it seems like an issue that only happens on the first run when you have to give permissions.

paredesivan, 2018-01-12:
@zerox1212 maybe yes

don, 2018-01-16:
I heard early versions of 7.0 and 7.1 had some problems but I haven't seen this. Does anyone have any stack traces or a way to reproduce this?

mardugas, 2018-01-16:
Hi Don,
how can I generate a suitable stack trace ?

don, 2018-01-16:
@mardugas connect your phone to your computer 
run `adb logcat` 
when you see an error cut and paste the relevant parts from the console

If you have [Android Studio](https://developer.android.com/studio/index.html) installed, you can run your app through Android Studio and filter to just show log messages from your app.

mardugas, 2018-01-18:
Here is the stack trace (I hope I selected the relevant parts). Any help is appreciated.

BLEPlugin: action = startScanWithOptions
...
BluetoothGatt: onClientConnectionState() - status=133 clientIf=5 device=D6:26:D2:4C:07:D8
BLEService: Connect fail.

[logcat7.1 extractBT.txt](https://github.com/don/cordova-plugin-ble-central/files/1644715/logcat7.1.extractBT.txt)

paredesivan, 2018-01-19:
function escanear(dispositivos, tiempoBusqueda) {

            console.log("entra en scanear");

            var defer = $q.defer();

            if (dispositivos == null) dispositivos = [];

            ble.scan(dispositivos, tiempoBusqueda, exito, falla);

            function exito(valor) {
                defer.resolve(valor);
                console.log("escaneados"+JSON.stringify(valor));
            }

            function falla() {
                defer.reject();
            }

            return defer.promise;

        }

esta es mi funcion de mi servicio y funciona bien

don, 2018-02-09:
@mardugas I'm not seeing an error just the generic connection failure message with status 133 gatt error. Can you use other Bluetooth apps on this same phone to connect to the device?

mardugas, 2018-02-09:
Other Bluetooth apps are working on the same phone (e.g. hands-free phone).
I could see the same error pattern (connection failure with BT) on a different smartphone (homtom, android 7)

crizhon, 2018-03-19:
I'm facing the same problem with Moto Z 7.x, some devices work fine and others do not connect, but work fine in Samsung s4 Mini 4.4 and Samsung J5 6.1, any solution?

don, 2018-05-07:
@crizhon there no easy answer. Try and restart the phone, or at least Bluetooth. The Bluetooth stack crashes on some devices. Enable the [Bluetooth HCI Snoop Log](https://blog.bluetooth.com/debugging-bluetooth-with-an-android-app) and debug in Wireshark.

I think these are device/os specific issues rather than issues with this plugin. If anyone is still having problem, please open a new issue and include the simplest example that causes the problem.

closed 2018-05-07 by don