don/cordova-plugin-ble-central #508: Memory Leak?

luisenrike, 2017-12-14:
So, I made an ionic app and I was monitoring the memory usage in Android Studio when I discovered that the allocated memory was always increasing. The allocated memory starts increasing when this line of code runs: `this.ble.startNotification(device.id, this.service, this.characteristic).subscribe(() => { });`, as you can see, I'm doing nothing in the `subscribe` handler.

I don't know if this is the expected behavior or maybe there is a memory leak somewhere.

[Video](https://streamable.com/uptzl)

don, 2018-01-16:
@luisenrike how many times are you calling start notification? Typically you call startNotification once per characteristic and keep the notification. This will use some memory. Is this causing your app to crash?

Can you provide me a simple app that demonstrates the leak?

luisenrike, 2018-01-17:
Hi @don, thanks for your reply. 

I only call `startNotification` once. The app is not crashing, but, after a while, it will drain the battery and the phone will heat up. The BLE device is sending a lot of data every 100 ms, so, maybe this is normal.

I made a simple ionic app:

https://github.com/luisenrike/ionicbleapp

https://streamable.com/ns78j

closed 2018-06-07 by don