don/cordova-plugin-ble-central #512: ble.enable and ble.startScan doesn't work

RdlP, 2017-12-21:
I have this code for enable function

```javascript
} else if(enscan.selectedIndex == 2) { // ble
            	alert("before");
                ble.enable(
                    function() {
                        alert("Bluetooth is enabled");
                    },
                    function(e) {
                        alert("The user did *not* enable Bluetooth: " + e);
                    }
                );
}
```

and similar for startScan (i have omitted code)

```javascript
ble.startScan([], function(device) {}, function(){});
```

The problem is that callbacks are never called. I am using `cordova-plugin-ble-central 1.1.4 "BLE"`, Android 4.0.4 on Samsung Galaxy Tab and I have this in config,xml

```
<platform name="android">
        <allow-intent href="market:*" />
        <preference name="android-targetSdkVersion" value="22" />
    </platform>
```
What could be the problem?
Thanks,

romanlutai, 2018-01-03:
I have the same issue on similar platform. Please, write if you have solved the problem

RdlP, 2018-01-09:
I continue having the same problem in my android 4.0.4. I have tried on Android 8.1 and the same code works

don, 2018-01-16:
@RdlP Bluetooth Low Energy doesn't work on Android 4.0.4. It was first available on 4.3, but not all 4.3 devices have the necessary hardware.

romanlutai, 2018-01-16:
@don, Android 6.0.1 on Samsung Galaxy S5 - does not work either. In general, the library works fine, but `ble.scan()` does not work.

don, 2018-01-16:
@romanlutai what happens when it "does not work"? Are you seeing errors starting scanning or just not getting any results.

Could this be related to permissions added in Android 6?

More details would be helpful.

romanlutai, 2018-01-16:
@don, it does not return any results. I don't think it is permissions issue, because all other functions are working. For instance, `ble.enable()` works well, and `ble.write()` also works with manually written MAC address.

homer-jay, 2018-03-01:
The scanning function doesn't work for me as well, at least not as expected. The callback is never called until the method `stopScan` is invoked.  
I'm on a Samsung tablet with OS 5.1.1, BLE is already enabled.

VolodymyrAndrushchak, 2018-03-01:
@homer-jay 
Have you tried to use scanWIthOptions?
Have you tried to specify the UUID of your BLE devices?

homer-jay, 2018-03-01:
@VolodymyrAndrushchak I was wrong. It probably works. I'm using the plugin through ionic which wraps the calls into RXjs observables. Apparently Observable doesn't kick in until you subscribe.

VolodymyrAndrushchak, 2018-03-01:
@homer-jay 
I also use this plugin in ionic project which I already use in production
It's my part of code for subscription

```
public startScan() {
   return Observable.from(this._ble.isEnabled())
     .catch(e => Observable.throw(new Error(‘Bluetooth is disabled. Please turn it on.’)))
     .flatMap(() => this._ble.startScanWithOptions(this.UUID, { reportDuplicates: true }))
     .filter(device => device.name === ‘my_name’ )
     .bufferTime(this._config.bufferTime)
     .flatMap(devices => Observable.from(devices).groupBy(d => d.id));
 }  
```

dheerajgupta27, 2018-04-03:
Getting "Error in Success callbackId: BLE924441947 : ReferenceError: failure is not defined" issue in cordova-plugin-ble-central plugin.

don, 2018-04-20:
@dheerajgupta27 that sounds like a different issue with Ionic1. I have an old Ionic1 sample application that might help you: https://github.com/don/ionic-ble

closed 2018-04-20 by don