don/cordova-plugin-ble-central #513: Issues with running multiple read commands very quickly on iOS

ewized, 2017-12-21:
So currently we are using the BLE Characteristics to get some data that changes very quickly. So we have multiple read command very quickly. The notification system is out of the question due to the way the firmware is implemented.

The problem is in this file: https://github.com/don/cordova-plugin-ble-central/blob/master/src/ios/BLECentralPlugin.m

The variable [here](https://github.com/don/cordova-plugin-ble-central/blob/master/src/ios/BLECentralPlugin.m#L46) is global so when multiple read commands are issued they are added to the dictionary [here](https://github.com/don/cordova-plugin-ble-central/blob/master/src/ios/BLECentralPlugin.m#L115) and the time it takes for a response on the line [here](https://github.com/don/cordova-plugin-ble-central/blob/master/src/ios/BLECentralPlugin.m#L564) it removes the Javascript callback function before it actually has time to process the next read command.

I'm not a `Objective C` nor `iOS` developer but I think you need some type of `MultiMap` to track multiple read commands and `.pop()` it when it receives a command, or have a better way to id the callback functions for the dictionary.

If you execute something like this:
```
window.ble.read(device_id, service_uuid, characteristic_uuid, success, failure);
window.ble.read(device_id, service_uuid, characteristic_uuid, success, failure);
window.ble.read(device_id, service_uuid, characteristic_uuid, success, failure);
window.ble.read(device_id, service_uuid, characteristic_uuid, success, failure);
```

Demo Log (As you can see it can read the first one but then loses the callback)
```
didUpdateValueForCharacteristic
bytesToString: MODE: 0 # our callback function
didUpdateValueForCharacteristic
didUpdateValueForCharacteristic
didUpdateValueForCharacteristic
```

titou71, 2018-01-05:
Hi,

I'm facing the same issue on Android. 
I have to read multiple message one the same characteristic UUID.

Do you have any idea to solve it ?

don, 2018-01-05:
@ewized Notifications are the proper way to get rapidly changing data. I'm not sure you can do this with reads. Generally reads need to be done serially. The APIs are asynchronous but you need to wait for a read to finish before the next one can happen. This is especially true on Android.

The best bet is to change your code to wait for the read to complete before issuing the next read. There may be ways to change the key and handle multiple callback, but that adds more complications. A change like that is best done in a fork first. If it makes sense and doesn't cause other problems it could be rolled back in.

I'm not even sure this will work for you. If I were you, I'd write a very simple iOS app to test app to make sure you can really read data that fast. If that works, I'd probably create a custom plugin that would read the data quickly and emit the results to JavaScript.

@titou71 on Android you need to wait for the first read to complete before the next read is issued. Android queue the commands for you.

ewized, 2018-01-06:
@don We can read the data that fast for our use case. I logged the output of the results to the console we are getting the data back. It's how I noticed the bug is part of the keys. We are going to make the changes in a fork to test it out but since I'm not an ObjectiveC developer it will take me some time.

closed 2018-06-07 by don