don/cordova-plugin-ble-central #514: problem when I try to writeWithoutResponse

paredesivan, 2018-01-08:
I have a problem. I scan, connect and write without without response. the first time all good. but then when I want to write it does not send the data, then I try to connect again and it tells me that I'm connected, so I try to write and it does not write. I also try to scan again, connect and write and do not send. I have to wait 20 seconds aprox  for it to work. I already change the scan time, change writewithoutresponse, i tried disconnect before the 20 seconds, but it does not work either. any ideas?

don, 2018-01-16:
Do you have some code you can share? Make sure the characteristic supports writeWithoutResponse. The peripheral determines if you should use write or writeWithoutReponse.

closed 2018-05-07 by don

paredesivan, 2018-05-22:
yes. the first time sends well

don, 2018-05-22:
If you use a tool like LightBlue Explorer or nRF Connect can you write multiple times?

Is the failure callback of the connect function being called? If so this means the peripheral disconnected. On Android sometimes the disconnect callback is delayed by 20 seconds after the disconnect occurs. 

It’s also possible there’s something wrong on the code. I’d write the simplest possible app to connect and write data to help debug this.

don, 2018-05-23:
I think the delay is in Android SDK since I don't see it on iOS. You can set a break point in the plugin's Java code, pull the power from a connected peripheral, and you should see a delay before Android tells you it's disconnected.

paredesivan, 2018-05-23:
i will try that