don/cordova-plugin-ble-central #515: Packet order not guaranteed on startNotification

xyberzues, 2018-01-10:
The slave is sending responses to master. The response is broken into 20 bytes packets and the last packet has a delimiter character that we use to check if the response is complete with assembling the packets together. I have noticed that sometimes the packet with the delimiter character will arrive before the reset of the packets in a response.

don, 2018-01-16:
I've always seen the packets arrive in order unless the peripheral is sending too fast. When sending notification faster then 30ms, I've seen iOS drop packets. 

My plugin gets the data from the OS. The notification should be acknowledged by the link layer. [This TI thread](https://e2e.ti.com/support/wireless_connectivity/bluetooth_low_energy/f/538/t/454246) mentions that order is guaranteed, but there was a bug in older versions.

I'd get a sniffer and wireshark on the connection and verify the notifications are coming from the peripheral in the correct order. If you have more info post it here.

xyberzues, 2018-01-23:
We used a sniffer and the packets are in the right order, could it be some race condition

don, 2018-02-09:
Are you seeing this on iOS or Android? If the packets are in the right order over the wire, it must be something in the OS.

zmoshansky, 2018-03-08:
@xyberzues 
AFAIK, all packets, regardless of type(write, write/without response, notification, indication), are acknowledged in BLE at the [physical layer](https://www.silabs.com/community/wireless/bluetooth/knowledge-base.entry.html/2016/05/02/acknowledged_vs_unac-u6eO).

I've also seen posts from both Nordic and Ti [devzones](https://devzone.nordicsemi.com/f/nordic-q-a/18482/is-there-some-sort-of-assurance-that-ble-packets-get-to-the-application-undamaged-and-in-the-correct-order) verifying that order should be maintained at physical layer (Dropped packets resulting in a re-transmission).

> Yes. At the end of every link layer packet there is a 24-bit CRC.
> Every packet will be acknowledged at the link layer, so if there isn't an acknowlegdement, it will be retransmitted.
-- Petter Myhre

That post goes on to say that iOS devices sometimes have TX data loss occur in the OS layer if the application writes too fast (Since BLE Hardware in phones is shared among many apps - Ie. Streaming to BT Headphones while communicating with a different BLE device.)

closed 2018-05-07 by don

timburke, 2019-01-07:
@goemen @zmoshansky Just a heads up in case you still are experiencing this issue.  I finally tracked it down to a race condition inside Cordova itself.  There are more details here:
https://github.com/don/cordova-plugin-ble-central/issues/625#issuecomment-451788534

The issue, as described occurs only on Android and is independent of the BLE stack itself.  It's just that Cordova doesn't guarantee that plugin results will be returned to javascript in the same order they were sent from the plugin in certain cases.