don/cordova-plugin-ble-central #516: Cannot disconnect and close the connection

haotangio, 2018-01-11:
`java.lang.NullPointerException: Attempt to invoke virtual method 'void org.apache.cordova.CallbackContext.sendPluginResult(org.apache.cordova.PluginResult)' on a null object reference`

This one looks like related to https://issuetracker.google.com/issues/37057260

Can someone have a look and fix it if possible? I'm trying to fix but may not be quick.

paredesivan, 2018-01-12:
try: wait for 100 ms before calling close. in the java file

don, 2018-01-16:
Unfortunately this is a bug on some devices. I started working on a fix but it broke other devices. I haven't had a lot of time to spend on this since my devices don't have the problem. Let me know if you're able to track this down or find a fix.

haotangio, 2018-01-16:
I tried several ways including waiting for 100ms before calling `close`. But it still doesn't work.

don, 2018-06-07:
Fixed in #592

closed 2018-06-07 by don