don/cordova-plugin-ble-central #517: Write commnad has 96 responses but notification success callback stops after first response

CaptainJeff, 2018-01-14:
I am writing to a characteristic whose success value returns 96 bytes however the notification successcall back stops after the first response. How do i get it to continue until the reponses are done?

I tried with write and writeWithoutResponse

don, 2018-01-15:
Are you seeing this on iOS or Android. Can you share a bit of code so I can see how you’re handling the notifications?

KingRial, 2018-01-17:
Could be related to https://github.com/don/cordova-plugin-ble-central/pull/447

I have a similar problem with Android 5.1.1 Sony Xperia

KingRial, 2018-01-17:
Could be related to https://github.com/don/cordova-plugin-ble-central/pull/447

I have a similar situation over Android 5.1.1 Samsung Xperia.

With that fix I was able to get and handle the errors correctly.

mburger81, 2018-02-14:
@CaptainJeff what do you intend with 96 responses, are you waiting for 96 responses in WRITE SUCCESS CALLBACK or are you waiting for 96 ARRAY BUFFERS in startNotification CALLBACKS?

on `write` and `writeWithoutResponse` success callback we get ALWAYS only an `OK` message.

CaptainJeff, 2018-02-14:
@mburger81 and @don 

Sorry I didn't respond earlier! I actually figured it out and it was my mistake and not the plugin. I was trying to get fancy and return the response in the callback of my function write command instead of on the startNotification callback and i was returning the response on the first piece of data returned.

@mburger81 the 96 responses are coming from a pedometer and returns the detailed data for the day in 96 reponses. But you guys were right and its returning all 96 in startNotification 

Sorry for that confusion

closed 2018-05-07 by don