don/cordova-plugin-ble-central #518: Don, I need you! (Feature/help request)

al-ec, 2018-01-15:
Hi Don,

Absolutely love this plugin and the examples you provided for Ionic.

Here's my current project:

Building a UART-esque serial app for BLE (specifically HM10 modules). I plan to use cordova-plugin-ble-central (or rather, ionic's "version") for my project, versus the Bluetooth Serial option (since it uses Classic for Android).

I've got it all working in Ionic, except as expected, my message drops after 20 characters. If you were to send from the peripheral:

`123456789012345678901`

You would receive back:

`1`

But that's to be expected, right?

My question is, can you provide an example of how I could add in a delimiter, or something that allows me to receive/send more than 20 bytes?

Or perhaps is there some way to modify Bluetooth Serial to force BLE for all platforms?

I'll take whatever resources you're willing to hand over. This is a little over my head.

Thanks so much! Great work on this plugin!

matiaspando, 2018-01-15:
I had the same problem and found a workaround splitting the message into chunks of 15 chars.

al-ec, 2018-01-15:
Hey @matiaspando!

Thanks for your response. I’ve heard about chunks but have never actually used them. I’m new to anything beyond plain JS.

Would you mind very graciously sharing some code samples of chunks being used for both reading and writing with this plugin?

Cheers!

al-ec, 2018-01-15:
Also, did some digging, and found this. Would it work?

https://github.com/don/cordova-plugin-ble-central/issues/234#issuecomment-340910307

Note I’m on the latest iOS, and have this 20 byte Rx/Tx limit, even though they mention that it is/was an Android problem?

don, 2018-01-15:
@al-ec This is likely a limitation of the HM-10. iOS and Android will usually transparently use use blob support to send larger characteristics. 

For some hardware we spilt the message into chunks of 20 characters before sending. I’ll see if I can find a sample for you.

al-ec, 2018-01-15:
Many thanks, @don. You're too good to us.

al-ec, 2018-01-15:
Also, @don please note that I don't think it's a limitation of the device. In other apps, including your bluefruitLE example, when I send an array of over 20 bytes from the computer (serial host) to the iOS device, it prints the full array in the HTML element, but splits it up into 20 character chunks with a "(line break) Received:" in-between.

So I think it's something I'm doing wrong here. Maybe it's because I'm trying to pass the raw data (from ble.startNotification) via ngZone?

Example:

`        this.ble.startNotification(this.peripheral.id, 'ffe0', 'ffe1').subscribe(
            dataGet => this.onMessageGet(dataGet),
            dataGet => this.onMessageGetFail(dataGet)
            ) `

`    onMessageGet(dataGet) {
        this.ngZone.run(() => {
        this.readMessage = this.bytesToString(dataGet);
    }
                       )}
   
   // if the promise returns the
   // alternate return...
onMessageGetFail(dataGet) {
console.log('Error message recieved from serial:' + dataGet);
this.ngZone.run(() => {
this.statusMessage = 'Error recieving data...'; })
}  `

And of course, on the debugging page to see the readout:

`      <ion-card>
    <ion-card-header>
        <ion-icon name="bluetooth" item-start></ion-icon> Data Recieved
    </ion-card-header>
    <ion-card-content>
    <pre>{{ readMessage }}</pre>
    </ion-card-content>
  </ion-card>`

Does my problem lay in how I'm showing this data?

don, 2018-01-16:
Here's how we broke up some data before sending from iOS/Android to a BLE device that only supported 20 byte characteristics

https://gist.github.com/don/ad3d6f444925c57d169baf54e262348f

al-ec, 2018-01-16:
Amazing, @don, thanks.

And for the reverse problem, with only receiving the last 20 bytes of the message (or the overflow from filled 20-byte Uint8Arrays), would I just work your solution in reverse?

Here's what I'm playing with currently (untested, probably will fail):

````
this.ble.startNotification(this.peripheral.id, 'ffe0', 'ffe1').subscribe(
      buffer => {
            let data = new Uint8Array(buffer);
             var groups = [], i;
             for (i = 0; i < data.length; i += 19) {
             groups.push(data.slice(i, i + 19)); }
                this.ngZone.run(() => {
                this.readMessage = JSON.stringify(groups); })
      });

````

don, 2018-01-16:
Here's a project that sends log files from an RFduino (nRF51) to Cordova. It chunks the file on the Arduino before sending. https://github.com/don/rfduino-logreader/blob/master/Arduino/RFduinoFileUploader/RFduinoFileUploader.ino

On the Cordova side, the notification just keeps receiving data. It looks for the token at the end of the file to know when the file is complete. https://github.com/don/rfduino-logreader/blob/master/www/js/index.js#L79-L88

al-ec, 2018-01-16:
Thanks again.

I've tried the innerHTML solution you used there, but I can't seem to find a way to make it work within the ngZone plugin I'm using. That's how I know it's possible to do over 20 bytes, since it worked in your bluefruitLE example!

Is there anything within the Ionic framework that lets you add to the existing contents of an ngZone instead of replacing the contents entirely?

al-ec, 2018-01-16:
@don, **YOU ARE THE BEST!**

Thanks to you, I solved my help request.

Here's the code that worked, for use in Ionic 3.x (for others who are struggling with a similar brick wall):

In TypeScript:

`logData = '  ';`
````
onConnected(peripheral) {  
    
    this.ngZone.run(() => {
      this.peripheral = peripheral;
              this.setStatus('Connected to ' + peripheral.name || peripheral.id);
        });
    
       this.ble.startNotification(this.peripheral.id, 'ffe0', 'ffe1').subscribe(
           buffer => {
          var s = this.arrayBufferToString(buffer);
            if (s.match('/')) {
            this.logData = s;
            this.ngZone.run(() => {
                this.readMessage = this.logData; })
        } else {
            this.logData += s;
            this.ngZone.run(() => {
                this.readMessage = this.logData; })
}
      });
    
        this.ble.read(this.peripheral.id, 'ffe0', 'ffe1').then(
      buffer => {
          var s = this.arrayBufferToString(buffer);
            if (s.match('/')) {
            this.logData = s;
            this.ngZone.run(() => {
                this.readMessage = this.logData; })
        } else {
            this.logData += s;
            this.ngZone.run(() => {
                this.readMessage = this.logData; })
}
      });

}
    
    clear(box) {
        this.logData = '';
            this.ngZone.run(() => {
                this.readMessage = ''; })
    }
````

And on your HTML document (for 'fancy' debugging):

````
      <ion-card>
    <ion-card-header>
        <ion-icon name="md-cloud-download" item-start></ion-icon>  Data Recieved  <button ion-button round outline (click)="clear(box)">Clear</button>
    </ion-card-header>
    <ion-card-content>
    <pre>{{ readMessage }}</pre>
    </ion-card-content>
  </ion-card>
````

al-ec, 2018-01-16:
Feel free to close this. Thanks!

don, 2018-01-16:
When DOM is changed from Bluetooth it's outside of the Angular lifecycle. With Ionic I wrap these updates in ngZone.run() so that Angular knows about the change and updates the UI.

	function onData(buffer) {
	  var s = arrayBufferToString(buffer);
	  console.log(buffer);
	  this.ngZone.run(() => {
	      someElement.innerHTML = someElement.innerHTML + s;
	  });
	}

closed 2018-01-16 by don