don/cordova-plugin-ble-central #519: show bluetooth setting in ios

sonicwong, 2018-01-17:
I found a plugin which able to show bluetooth setting in iOS:
https://github.com/guyromb/Cordova-open-native-settings

Would you add this future?
Thanks!

don, 2018-01-18:
@sonicwong that looks like a great plugin. The big question is, does apple approve of using 
`prefs:root=Bluetooth` or is it a private URI? Please add to this issue if you see more info. I'd need to be sure apps wouldn't get rejected. 

In the mean time you should be able to install [Cordova-open-native-settings](https://github.com/guyromb/Cordova-open-native-settings) along side this plugin.

don, 2018-01-18:
People are reporting AppStore rejections using this https://gist.github.com/johnny77221/bcaa5384a242b64bfd0b8a715f48e69f (link found via https://github.com/phynet/iOS-URL-Schemes#bluetooth)

sonicwong, 2018-01-18:
@don thanks for your info. I close this issue first.

closed 2018-01-18 by sonicwong

don, 2018-01-18:
@sonicwong I talked to some more people about this. Unfortunately it looks like bluetooth preferences  is considered a private API.