don/cordova-plugin-ble-central #520: ble.connect not working iOS 11.2.2

hornej, 2018-01-17:
I have tried using both the MAC address and UUID of the peripheral and have been unable to connect. I get "Could not find peripheral ..." with either the UUID or MAC address. 

However, using the same code I am able to connect and write on Android with no issues. 

In this example I am trying to connect with the mac address
Code:
```
ble.scan([], 5, (peripheral) => {
                                console.log(peripheral);
                                     console.log('so close!!!');
                                     console.log(peripheral);
                                     console.log(peripheral.id);
                                     
                                if (macs[peripheral.id]) {
                                     console.log('found you!');
                                     console.log('hello????');
                                    ble.connect(macs[peripheral.id], (success) => {
                                                
                                    console.log('connected');
                                        console.log('attempt to write');
                                        ble.writeWithoutResponse(macs[peripheral.id], 'd304dbd9-6fdc-4bf3-a617-e015edd87ed5', '5099cbc8-a71f-4292-8158-bf4f25ae9948', 'GPIO_ON', (result) => {
                                            `console.log('written');`
```

Debug Statements:

> 2018-01-17 14:12:21.489584-0700 Piero[2776:1053935] {"id":"408E326D-CB8C-3FF9-5FDA-E5925FF2D7B1","rssi":-47,"advertising":{"kCBAdvDataIsConnectable":true},"name":"raspberrypi"}
> 2018-01-17 14:12:21.489634-0700 Piero[2776:1053935] so close!!!
> 2018-01-17 14:12:21.489669-0700 Piero[2776:1053935] {"id":"408E326D-CB8C-3FF9-5FDA-E5925FF2D7B1","rssi":-47,"advertising":{"kCBAdvDataIsConnectable":true},"name":"raspberrypi"}
> 2018-01-17 14:12:21.489692-0700 Piero[2776:1053935] 408E326D-CB8C-3FF9-5FDA-E5925FF2D7B1
> 2018-01-17 14:12:21.489715-0700 Piero[2776:1053935] found you!
> 2018-01-17 14:12:21.489737-0700 Piero[2776:1053935] hello????
> 2018-01-17 14:12:21.489761-0700 Piero[2776:1053935] connect
> 2018-01-17 14:12:21.489818-0700 Piero[2776:1053935] Could not find peripheral B8:27:EB:05:BC:89.
> 2018-01-17 14:12:21.491872-0700 Piero[2776:1053935] connection error
> 2018-01-17 14:12:21.491931-0700 Piero[2776:1053935] Could not find peripheral B8:27:EB:05:BC:89.
>

al-ec, 2018-01-18:
I’m on iOS and have no issues. We need much more info. What’s your peripheral? Can you post your ble.connect code?

al-ec, 2018-01-18:
Edit- Saw you included your code. Sorry!

al-ec, 2018-01-18:
Try creating different methods for each ble.method and see if that works.

don, 2018-01-18:
The log file shows MAC addresses for the connection "Could not find peripheral B8:27:EB:05:BC:89."

On iOS you need to use the UUID. iOS generates a UUID and hides the MAC address. Note that the UUID id phone specific. When your phone scans and sees your device B8:27:EB:05:BC:89 it gets one UUID. A second iPhone will see that same device and get a different UUID. It's Apple doing this not me :)

You can use [LightBlue Explorer](https://itunes.apple.com/us/app/lightblue-explorer/id557428110?mt=8) to connect to a device and get the UUID on your phone. Ultimately you need a better solution like finding devices by service id.

hornej, 2018-01-23:
@don I got it working thank you!! I was getting confused with the characteristic_uuid and service_uuid. Also I needed to add the stringToBytes function as well for the data 

You're right that I need to find a better solution. Does this plugin allow for finding devices by service id on iOS?

closed 2018-01-28 by hornej