don/cordova-plugin-ble-central #521: ble.connect not working on Android 6.0

h-brs-michael, 2018-01-18:
Hey,
ble.connect wont work on my huawei p8 lite 2015 phone with Android 6.0 with the error message "Peripheral Disconnected" in 2 situations.

Situation 1: Localtion-Services are enabled:
It does not matter, if the phones uses gps or wireless networks for the location services.
There is no way to connect to a peripheral, when the location services are enabled.

Situation 2: After a device restart:
disabling and enabling the bluetooth sensor fixes this situation.

The Android-Log from OnConnectionStateChange-Callback:
```
D/Peripheral: onConnectionStateChange status 133, newState 0
```
In both situations a ble scan is running in the background:
```
01-18 14:47:53.686 12031-12067/? I/Bluetooth: BtGatt.GattService:onScanResult() - address=98:07:2D:3A:C8:1A, rssi=-93
01-18 14:47:53.686 12031-12067/? I/Bluetooth: BtGatt.GattService:need phone set:requiresLocationEnabled=false, locationEnabled=true
01-18 14:47:53.686 12031-12067/? I/Bluetooth: BtGatt.GattService:need ble device set:hasPeersMacAddressPermission=false, hasLocationPermission=true, hasPermission:true
01-18 14:47:53.686 12031-12067/? I/Bluetooth: BtGatt.GattService:matchesFilters  return false
```

The scan was not started from my App and does not stop when ble.stopScan is called.
This might be the problem, because a connect does not work when a ble scan is still running.

Whenever a connect to a peripheral does work, there is no ble scan running in the background.

I'll check it with a different Android device soon.

h-brs-michael, 2018-01-24:
I tested it with an OnePlus 3t and the errors did not occur. The problem might be Huawei specific.

don, 2018-02-09:
@h-brs-michael it sounds like a problematic device or Android version.

ohdonpiano, 2018-02-28:
Same problem here, but it's not cordova related, I'm developing with native Android API and I have the same behaviour with Huawei P8 Lite.
I'm still searching for a solution

backraw, 2018-03-12:
I'm seeing that issue aswell. I have somewhat come closer to a fix by changing GATT autoConnect to *true* instead of *false* for lines 72 and 74 in Peripheral.java:
```java
if (Build.VERSION.SDK_INT < 23) {
    gatt = device.connectGatt(activity, true, this);
} else {
    gatt = device.connectGatt(activity, true, this, BluetoothDevice.TRANSPORT_LE);
}
```

The first time I try to connect to the peripheral with a Samsung Galaxy J5 (Android 6.0), it says **Peripheral Disconnected**, also my peripheral says/thinks it still is connected. The second time I try to connect it works and I get the peripheral data. But it's not guaranteed that it works in any case and for any Android version/device.

However, I can say for certain that the same app (with an unchanged Peripheral.java) with an ASUS P023 (Android 5.0) works without any problem. That is, connecting to the peripheral and getting its data, at least. I was experimenting with *Qt for Android* before encountering Cordova/Ionic, and I was experiencing the **same exact issue**. So I'm like 90% sure that the issue is somewhere in the Android 6 platform sources and is not (directly) related to this plugin.

So, has anyone found some info in some Android bug reports or something similar?

don, 2018-05-07:
Closing this issue since it sounds like a device specific problem.

closed 2018-05-07 by don