don/cordova-plugin-ble-central #523: ios Connection error

hehongfe, 2018-01-23:
/*my code*/
     this.ble.connect(device.id).subscribe(
            peripheral => {
              this.isContent=true;
              this.onConnected(peripheral,serviceUUID,characteristicUUID,characteristicUUID_L,data)
      },(err) => this.onDeviceDisconnected(err));

     onConnected(peripheral,serviceUUID,characteristicUUID,characteristicUUID_L,data){
        this.goStartNotification(peripheral,serviceUUID,characteristicUUID_L);
        let ArrayBuffer=this.myStringByte(data);
        //写入蓝牙数据
        this.goWrite(peripheral,serviceUUID,characteristicUUID,ArrayBuffer);
    }

goWrite(device,serviceUUID,characteristicUUID,buff){
    this.ble.write(device.id,serviceUUID,characteristicUUID,buff).then((r:any)=>{
          console.log(r);// 返回：ok
        },(err)=>{
          this.hideLoading();
          console.log(err)
      })
  }

{"errorDescription":"The connection has timed out unexpectedly.","id":"3116550A-3F57-256B-A27C-45CAC62009B1","name":"XDYS-991108961E31","errorMessage":"Peripheral Disconnected"}

hehongfe, 2018-01-23:
cordova-plugin-ble-central ios error

don, 2018-02-09:
@hehongfe based on your error, the peripheral is disconnecting. Does this happen after you write? Maybe the data you're sending is causing it to disconnect? 

If you send the same data using [LightBlue Explorer](https://itunes.apple.com/us/app/lightblue-explorer/id557428110?mt=8) does it also disconnect?

It looks like you're using Ionic, you might want to check out my Ionic sample code https://github.com/don/ionic-ble-examples

closed 2018-05-07 by don