don/cordova-plugin-ble-central #526: ble bond

amitvikram1985, 2018-01-24:
Hi,
I am using this plugin in one my application for both android and ios.
I have one issue regrading ble bond. I am not able to set bond because not able any function to set bond 
using this plugin.

Please help me to set bond,This is important for  me.

Amit.

hermannle, 2018-02-05:
Hi,
I've got the same problem as Amit. Bonding is a crucial security feature for industrial applications. It decides whether to use this framework or not.

don, 2018-02-09:
@amitvikram1985 if you try and access a characteristic that requires elevated encryption your phone should try and prompt you to begin bonding.

@hermannle if you need bonding for a project, I'm happy to talk. Features usually added when I need them for a customer's project or someone hires me to add them.

don, 2018-06-13:
Bonding currently needs to happen outside of the app in Android. Android now has ble.bondedDevices() to get the bonded devices. I create new issue #605 to track bonding.

closed 2018-06-13 by don