don/cordova-plugin-ble-central #527: Add a method to test Location Service

hervehobbes, 2018-01-25:
Hello,

May I can suggest to add a method to test if the Location Service (Android) is activated please?

bye

don, 2018-02-09:
@hervehobbes this is a good idea. Can you submit a pull request?

Domvel, 2018-05-29:
The scan method should request the location. Do not add a separate API method. Becuase this plugin should do this, not the app developer. Currently I'm using the ionic plugin "Diagnostics". There is a method to check and request the location. 
- Requests the permission. (check it by disable the permission in the settings)
- Check if location is enabled.

don, 2018-05-29:
@Domvel that's a good idea. The plugin should return an error if location is disabled. The probably belongs right after the [permission check](https://github.com/don/cordova-plugin-ble-central/blob/82a93ba9022f298b71db7f066ec34331a8e610e7/src/android/BLECentralPlugin.java#L506)

Domvel, 2018-05-29:
You can also request the location permission. Then the user will see a dialog with **Deny** and **Allow** Buttons. Also a checkbox with **Never ask again** on deny. This will switch the toggle button in the permission settings of this app. 

We have to handle the denied permission AND disabled location in the settings. 
On authorization request you will get a promise. You can wait until the user has decided. For example with then or async-await. But finally if the location still denied or is not enabled, the BLE plugin scan method (maybe other methods or on general usage?) has to throw an error "Location is not available."

I have no idea why is location required for bluetooth scan. Maybe only for data collection? 😄

don, 2018-05-29:
@Domvel the code already requests the location permission today. Bluetooth beacons can be used for location, so scanning requires location permission.

commit 58a3d2c7 (2018-06-08): Location services check (#595)

scan returns error if location services are disabled
fixes #527
closed 2018-06-08 by don