don/cordova-plugin-ble-central #528: Android 8 PHY settings

shibbs, 2018-01-31:
Hi There,

We're having a lot of issues with Android 8 (Oreo) phones and ble scans. Basically our BLE scans are returning nothing, but customers are claiming they can see our device in their phone's bluetooth scan. 

It seems like some of the issues are typical Android hardware/OS problems, like in [this forum](https://productforums.google.com/forum/#!topic/nexus/QxcbWBsnvWc). In that case we're basically hosed. 

However, one potential issue could be that Android 8 has added some PHY settings to make it compatible with BLE 5. From reading [this article](https://blog.bluetooth.com/what-bluetooth-developers-should-know-about-android-o) it seems like if the PHY was not set to PHY_LE_1M_MASK the OS will be operating in BT 5 mode only, and will be unable to interact with 4.x devices. 

Has anyone played around with this? I'll try to take a shot at playing with that setting in the next few days if not, but as usual I can't replicate the issues some users are having. 

Thanks,
Shibbs

kaddyadriano, 2018-02-20:
I am having the same issue. @don can you help us please!

cairinmichie, 2018-03-02:
I can confirm that scans are working with the Nexus 6P on android 8.1. So more likely that it's a OS problem.

VolodymyrAndrushchak, 2018-03-02:
I use this plugin in production for my Ionic project.
Its also perfectly works on Nokia and OnePlus with Android 8 (Oreo)
Have you tried on other Android devices with Oreo?
Be sure that your Android studio has the latest updates!

don, 2018-05-07:
@shibbs @kaddyadriano are you still experiencing this issue or have android updates fixed it?

don, 2018-06-13:
@kaddyadriano @shibbs I'm going to assume that the Android patches fixed this or you found another solution. If it's still a problem LMK and I'll re-open this.

closed 2018-06-13 by don