don/cordova-plugin-ble-central #529: Unable to turn on temperature notification ( or read it )

m-menon, 2018-02-01:
hi,

I have been able to scan/connect to a TI beacon and print the complete peripheral data. As a matter of fact I have even been able to turn on the battery voltage notification that works perfectly great! But getting the temperature reading has proved to be a challenge. The manufactured mentioned that it is possible to read it from the advertisement data. And I kept gathering this data only to find that it doesn't change at all ( except for the part containing the battery voltage - which I can decode from this string ). So now that I have no chance of reading the temperature from the advertisement string, I tried connecting as mentioned and then turn on the notification. But I just can't get it to work. Any pointers would be so helpful. Thanks a zillion. I am attaching the peripheral data from the beacon

```
{
  "name": "Micro Temp Sensor",
  "id": "50:65:83:xx:xx:xx",
  "advertising": {
    
  },
  "rssi": -70,
  "services": [
    "1800",
    "1801",
    "180a",
    "c050",
    "c090",
    "f000ffc0-0451-4000-b000-000000000000"
  ],
  "characteristics": [
    {
      "service": "1800",
      "characteristic": "2a00",
      "properties": [
        "Read"
      ]
    },
    {
      "service": "1800",
      "characteristic": "2a01",
      "properties": [
        "Read"
      ]
    },
    {
      "service": "1800",
      "characteristic": "2a02",
      "properties": [
        "Read",
        "Write"
      ]
    },
    {
      "service": "1800",
      "characteristic": "2a03",
      "properties": [
        "Write"
      ]
    },
    {
      "service": "1800",
      "characteristic": "2a04",
      "properties": [
        "Read"
      ]
    },
    {
      "service": "1801",
      "characteristic": "2a05",
      "properties": [
        "Indicate"
      ],
      "descriptors": [
        {
          "uuid": "2902"
        }
      ]
    },
    {
      "service": "180a",
      "characteristic": "2a25",
      "properties": [
        "Read"
      ]
    },
    {
      "service": "180a",
      "characteristic": "2a26",
      "properties": [
        "Read"
      ]
    },
    {
      "service": "180a",
      "characteristic": "2a29",
      "properties": [
        "Read"
      ]
    },
    {
      "service": "c050",
      "characteristic": "c051",
      "properties": [
        "Read"
      ]
    },
    {
      "service": "c050",
      "characteristic": "c052",
      "properties": [
        "Read"
      ]
    },
    {
      "service": "c050",
      "characteristic": "c053",
      "properties": [
        "Read"
      ]
    },
    {
      "service": "c050",
      "characteristic": "c054",
      "properties": [
        "Read"
      ]
    },
    {
      "service": "c050",
      "characteristic": "c055",
      "properties": [
        "Read"
      ]
    },
    {
      "service": "c050",
      "characteristic": "c056",
      "properties": [
        "Read",
        "Write"
      ]
    },
    {
      "service": "c050",
      "characteristic": "c057",
      "properties": [
        "Read"
      ]
    },
    {
      "service": "c050",
      "characteristic": "c058",
      "properties": [
        "Read"
      ]
    },
    {
      "service": "c050",
      "characteristic": "c059",
      "properties": [
        "Read",
        "Write"
      ]
    },
    {
      "service": "c050",
      "characteristic": "c05a",
      "properties": [
        "Notify"
      ],
      "descriptors": [
        {
          "uuid": "2902"
        }
      ]
    },
    {
      "service": "c050",
      "characteristic": "c05b",
      "properties": [
        "Read",
        "Write"
      ]
    },
    {
      "service": "c050",
      "characteristic": "c05c",
      "properties": [
        "Read",
        "Write"
      ]
    },
    {
      "service": "c050",
      "characteristic": "c05d",
      "properties": [
        "Read",
        "Write"
      ]
    },
    {
      "service": "c090",
      "characteristic": "c091",
      "properties": [
        "Notify"
      ],
      "descriptors": [
        {
          "uuid": "2902"
        }
      ]
    },
    {
      "service": "c090",
      "characteristic": "c092",
      "properties": [
        "Read",
        "Write"
      ]
    },
    {
      "service": "c090",
      "characteristic": "c093",
      "properties": [
        "Read",
        "Write"
      ]
    },
    {
      "service": "c090",
      "characteristic": "c094",
      "properties": [
        "Read",
        "Write"
      ]
    },
    {
      "service": "c090",
      "characteristic": "c095",
      "properties": [
        "Read",
        "Write"
      ]
    },
    {
      "service": "f000ffc0-0451-4000-b000-000000000000",
      "characteristic": "f000ffc1-0451-4000-b000-000000000000",
      "properties": [
        "WriteWithoutResponse",
        "Write",
        "Notify"
      ],
      "descriptors": [
        {
          "uuid": "2902"
        },
        {
          "uuid": "2901"
        }
      ]
    },
    {
      "service": "f000ffc0-0451-4000-b000-000000000000",
      "characteristic": "f000ffc2-0451-4000-b000-000000000000",
      "properties": [
        "WriteWithoutResponse",
        "Write",
        "Notify"
      ],
      "descriptors": [
        {
          "uuid": "2902"
        },
        {
          "uuid": "2901"
        }
      ]
    }
  ]
}
```

don, 2018-02-09:
Unfortunately you're going to need details from the manufacturer on how to use this. It's possible you need to write to a configuration characteristic to enable the temperature readings before you receive notifications or they are broadcast in the advertising data. Take a look at the TI SensorTag documentation and see if your sensor has similar behavior http://processors.wiki.ti.com/index.php/CC2650_SensorTag_User's_Guide#IR_Temperature_Sensor

Alternately if there's an app that works with the hardware, enable Android's Bluetooth HCI snoop log, btsnoop_hci.log, in the developer settings. Run the app, then load the log into Wireshark to see how they enable the thermometer.

m-menon, 2018-02-09:
FANTASTIC! Thanks a ton for the pointers. That should help me sort it out. And wonderful work you've done with this plugin. It was the only one  (of the few that I tried) that worked very reliably every time. Thanks once again :100:

m-menon, 2018-02-12:
oops, a minor hiccup I guess. I had some trouble with BT HCI snoop log. It wasn't creating a log file ( I tried 3 different devices and made sure that there was a SD card inserted - cos the snoop log thingy seemed to prefer to write to SD card from what I read ). Then I ran [HCI Debugger v1.84](https://play.google.com/store/apps/details?id=com.github.akinaru.hcidebugger&hl=en) and I could gather some information about the data being exchanged. But the best bet I had was with Android app called [LightBlue](https://play.google.com/store/apps/details?id=com.punchthrough.lightblueexplorer&hl=en). At last I can see some bytes being written ( 2 bytes ) that seem to turn on the sensor. But I am a little confused because it is writing to a descriptor.  Is it like turning on the sensor and the notification at the same time. If I could have a hint that'll be cool. Here's a log from [LightBlue](https://play.google.com/store/apps/details?id=com.punchthrough.lightblueexplorer&hl=en) ( which seems to enable the sensor perfectly fine ). Wondering how I could mimic that behavior in the correct way. 
[image]

```
Tue Feb 13 02:04:53 GMT+08:00 2018
Success: Connected to device.
MAC address:50:65:83:10:E3:B2

Tue Feb 13 02:04:54 GMT+08:00 2018
Success: Successfully discovered services. 6 service(s) found.

Tue Feb 13 02:04:54 GMT+08:00 2018
Status: Reading from characteristic 00002a29-0000-1000-8000-00805f9b34fb.

Tue Feb 13 02:04:54 GMT+08:00 2018
Success: Read from characteristic
00002a29-0000-1000-8000-00805f9b34fb.

Tue Feb 13 02:04:54 GMT+08:00 2018
Status: Reading from descriptor 00002901-0000-1000-8000-00805f9b34fb.

Tue Feb 13 02:04:54 GMT+08:00 2018
Success: Read from descriptor: 00002901-0000-1000-8000-00805f9b34fb. Data: Img Identify

Tue Feb 13 02:04:54 GMT+08:00 2018
Success: Read from descriptor
00002901-0000-1000-8000-00805f9b34fb.

Tue Feb 13 02:04:54 GMT+08:00 2018
Status: Reading from descriptor 00002901-0000-1000-8000-00805f9b34fb.

Tue Feb 13 02:04:54 GMT+08:00 2018
Success: Read from descriptor: 00002901-0000-1000-8000-00805f9b34fb. Data: Img Block

Tue Feb 13 02:04:54 GMT+08:00 2018
Success: Read from descriptor
00002901-0000-1000-8000-00805f9b34fb.

Tue Feb 13 02:05:17 GMT+08:00 2018
Status: Writing 2 bytes to descriptor 00002902-0000-1000-8000-00805f9b34fb

Tue Feb 13 02:05:17 GMT+08:00 2018
Success: Written to descriptor
00002902-0000-1000-8000-00805f9b34fb
Data:
�

Tue Feb 13 02:05:18 GMT+08:00 2018
Status: Received indication/notification on characteristic change:
0000c05a-0000-1000-8000-00805f9b34fb
Data:
R �

Tue Feb 13 02:05:20 GMT+08:00 2018
Status: Received indication/notification on characteristic change:
0000c05a-0000-1000-8000-00805f9b34fb
Data:
RN�

Tue Feb 13 02:05:22 GMT+08:00 2018
Status: Received indication/notification on characteristic change:
0000c05a-0000-1000-8000-00805f9b34fb
Data:
RN|

Tue Feb 13 02:05:24 GMT+08:00 2018
Status: Received indication/notification on characteristic change:
0000c05a-0000-1000-8000-00805f9b34fb
Data:
RNv
[... 96 lines cut ...]
0000c05a-0000-1000-8000-00805f9b34fb
Data:
Q�T

Tue Feb 13 02:05:52 GMT+08:00 2018
Status: Received indication/notification on characteristic change:
0000c05a-0000-1000-8000-00805f9b34fb
Data:
R R

Tue Feb 13 02:05:54 GMT+08:00 2018
Status: Received indication/notification on characteristic change:
0000c05a-0000-1000-8000-00805f9b34fb
Data:
R P

Tue Feb 13 02:05:56 GMT+08:00 2018
Status: Received indication/notification on characteristic change:
0000c05a-0000-1000-8000-00805f9b34fb
Data:
R O

Tue Feb 13 02:05:57 GMT+08:00 2018
Warning: Disconnected from and closed GATT for device.
```

don, 2018-02-12:
When you subscribe for notifications (or indications) a value is written to the 2902 descriptor. The plugin writes to the descriptor when you subscribe.

The hci log is usually written to the sdcard directory but should work find without a card. Use adb shell to look at your phone. I think I've documented the steps in more detail on other issues in this repo.

m-menon, 2018-03-10:
hi @don ... I think I have no other problem in the recent years that has taken me this long. I just can't seem to get the temperature readings. I made a complete app using this plugin and can do almost everything from connecting to writing to characteristics, reading, enabling notifications etc. Would you have a moment to look at my [SO post](https://stackoverflow.com/questions/49206882/cant-get-temperature-reading-out-of-ble-beacon-at-my-wits-end-now-this-need)? I have posted almost every single detail including a minimal communication capture using wireshark and a ble sniffer. To not make the log unduly log, all I did was sniff it just long enough to capture the connect, record a few temperature readings and disconnect. It could eventually be a tiny thing that I could be missing. Unfortunately this is my first experience with bluetooth low energy so I am completely stalled at this point.