don/cordova-plugin-ble-central #530: scan() and startScan(); doesn't work at all on either iOS or Android.

rooz121359, 2018-02-04:
Hello,

Thank you for this plugin. 

However, I can't seem to make this work at all!

I'm trying to use the following code to start a scan but I never see anything being scanned even though I know there are BLE devices only 10 cantemeters away of my app device!

Tried this:

rooz121359, 2018-02-04:
Okay, I seem to have made it work on iOS but still not scanning on Android.

I use this code which started working on iOS:

```
ble.isEnabled(
  function(){
    // Bluetooth is enabled
	console.log('Bluetooth is enabled');
	
	ble.startScan([], function(device) {
    console.log(JSON.stringify(device));
    }, function(){

    });
	
	
  },
  function(){
    // Bluetooth not yet enabled so we try to enable it
    ble.enable(
      function(){
        // bluetooth now enabled
		console.log('bluetooth now enabled');
      },
      function(err){
        alert('Cannot enable bluetooth');
      }
    );
  }
);
```

I have this in my config.xml file:

<plugin name="cordova-plugin-ble-central" source="npm" />

<preference name="pgb-builder-version" value="2" />
<preference name="phonegap-version" value='cli-7.1.0' />
<preference name="android-build-tool" value="gradle" />
<preference name="android-minSdkVersion" value="23" />
<preference name="android-targetSdkVersion" value="23" />

rooz121359, 2018-02-06:
Anyone?

don, 2018-02-09:
This sounds like an error in you application code. Look for errors in the log file with 'adb debug' and use chrome://inspect to find errors in the web view.

Try this sample app on your devices https://github.com/don/ble-scan

rooz121359, 2018-02-09:
@don cheers don. appreciate the reply.

I will give that ago to see if that helps. Just out of interest, have you tried that code on both iOS and Android?

don, 2018-02-09:
I haven’t run that example code lately but it was tested on iOS and Android. I regularly use other apps written with this plugin on both iOS and Android.

closed 2018-02-12 by don