don/cordova-plugin-ble-central #531: Issues searching for and connecting iOS

hornej, 2018-02-08:
I am trying to build something that involves multiple phones and multiple raspberry pi's. The pi's are acting as BLE beacons and the iOS app is scanning for, connecting, and writing to these raspberry pi's.  

The problem I am having is identifying these devices. 
- iOS hides the MAC address and creates a UUID instead so I can't search using MAC address.
- I tried naming the raspberry pi something unique and searching for the peripheral name. However this does not work because iOS will cache bluetooth names only after it connects. This means that the device would be seen as 'raspberrypi' or 'unknown' and iOS wouldn't recognize the unique name I gave it until after it connects. 

I need to be able to discover my raspberry pi's with some universal id so that I can scan and get the UUID that iOS creates for them so that I can write to them. The current solution I am working with is connecting to all nearby peripherals and trying to write to them which is not ideal. 

I am very confused because I feel like this is a common thing to do with BLE and there should be a way to do it. Thanks for the help

The code to scan, connect and write
`result.beacons.forEach((beacon) => {
                                                                if (beacon.proximity === 'ProximityImmediate' || beacon.rssi >= -65) {
                                                                console.log('attempt to connect');
                                                                console.log('initiate scan');
                                                                ble.scan([], 5, (peripheral) => {
                                                                         console.log("periphs", JSON.stringify(peripheral,null,2))
                                                                         //console.log("UUID's", beacon.uuid)
                                                                         //if (peripheral.name == 'josh') { //this idea did not work. see explanation
                                                                         if (beacon.uuid == 'uuid_hardcoded_into_rpi_bleno_script') {
                                                                         console.log('uuid found!');
                                                                         console.log(beacon.uuid);
                                                                         let perph = peripheral.id;
                                                                         console.log(perph);
                                                                         console.log('perph saved');
                                                                         
                                                                         ble.connect(perph, (success) => {
                                                                                     console.log('connected');
                                                                                     console.log('attempt to write');
                                                                                     ble.writeWithoutResponse(perph, 'D304DBD9-6FDC-4BF3-A617-E015EDD87ED5', '5099CBC8-A71F-4292-8158-BF4F25AE9948', stringToBytes('GPIO_ON'), (result) => {
                                                                                                              console.log('written');`

don, 2018-02-09:
If you are using iBeacons on the Raspberry Pi you need to use CoreLocation on iOS. Beacons are BluetoothLE, but iOS hides this an make you use other APIs. Try https://github.com/petermetz/cordova-plugin-ibeacon if you want to stick with Cordova.

If you're creating your own beacons, you can broadcast names, advertising info, characteristic values etc. iOS caches aggressively so if you're not sending services changes when changing the peripheral, toggle bluetooth on and off in the settings to avoid cached info.

closed 2018-05-07 by don

gtroja, 2018-08-17:
We were having problems with cached names in ios because our ble devices change names and our Iphone app were unable to find the device by its new name. One solution we found was instead of using "name" field in the response of scanning to check if it was the one we were looking for, we used "advertising.kCBAdvDataLocalName"