don/cordova-plugin-ble-central #534: scan() and startScan(); runtime error on Android.

Kleinast, 2018-02-10:
Ionic Framework: 3.9.2
Ionic App Scripts: 3.1.8
Angular Core: 5.0.3
Angular Compiler CLI: 5.0.3
Node: 8.9.4
OS Platform: Windows 10
Navigator Platform: Win32
User Agent: Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/64.0.3282.140 Safari/537.36

My code on ionic:
app.module.ts
```
import { BLE } from '@ionic-native/ble';

  providers: [
    BLE
  ]

```
home.ts
```
import { BLE } from '@ionic-native/ble';

constructor(
    private BLE: BLE
  ) {
  }

ionViewDidEnter() {
    this.BLE.startScan([]).subscribe(device => {
     console.log(JSON.stringify(device));
   });

   setTimeout(() => {
      this.BLE.stopScan();
    }, 5000);
  }

```

Did I do something wrong ? or is it a not normal it didn't work ?

Thanks in advance

don, 2018-02-12:
You didn't provide any details about the error.
See the https://github.com/don/ionic-ble-examples for working Ionic examples.

closed 2018-02-12 by don

valenzuela21, 2018-07-22:
Hello Problem Scan  Bluetooth Ble Native 
[image]
Ionic

valenzuela21, 2018-07-22:
Problem with the image above can someone help me!
I followed all the steps to the letter and nothing.
https://don.github.io/slides/2017-07-20-ionic-madison/#/

valenzuela21, 2018-07-22:
I am communicating communicating with Bluefruit LE - Bluetooth Low Energy (BLE 4.0) - nRF8001 Breakout - v1.0.
I will have problems with UUID services?