don/cordova-plugin-ble-central #535: Question or Feature Request: Notifications with more then 8bytes

mburger81, 2018-02-12:
We use your cordova lib in IONIC to listen for BLE notifications. It is working very good but now we are behind a implementation problem.

We send a 8byte command to a BLE device which has a 32byte acknowledge which we should get over BLE advertising.  On your `startNotification` listener  we get always 8bytes, so it seems we get 4 responses with always 8 bytes.
So we are not an expert on bluetooth and I'm not sure if BLE can do an advertising with more then 8bytes and we should expect 4 responses. But in this case the problem is how can we be sure the repsonses we see in log are in the right sequences. 

I saw some implementations on java where you configure a response length and max timeout. This is a function like `sendAndReceive(cmd, length)` where they send the command and wait for the response with max default timeout

So the question is, how can we do with your plugin a implementation like this, is this a default behavior for BLE or is this something really customized for the HEALTH CARE device we try to implemented.
In this way BLE does not support sending parallel multiple commands for multiple responses, but it is a SERIAL protocol.

What's your opinion?

Thx
Michael

don, 2018-02-12:
The Android Bluetooth APIs are asynchronous but the commands must be executed serially. The plugin code queues the commands and waits for a response before sending the next command.

Your question about notifications and 8 vs 32 byte responses are device specific. I'm not sure what device you have or how it works. Some device are limited to 20 bytes pre characteristic, so it is possible the device just breaks up responses. You should receive the notifications in order.

mburger81, 2018-02-12:
Hey thx, what do you intend with "The plugin code queues the commands and waits for a response before sending the next command."
Does this the write method sending a command? Does the succes callback returning the next advartisinf for GATT? 
I don't think so, so I'm no sure what do you intend but it sounda good :)

mburger81, 2018-02-14:
The write method returns only `OK` in success callback, so at the moment we don't  understand what do you intend with 

> The plugin code queues the commands and waits for a response before sending the next command.

don, 2018-05-07:
The commands are queued on Android because of the way Android works. The APIs look like they are asynchronous, but they need to be called serially. (You'll need to ask Google why they did this.)

When you write, you'll get a success callback if the write succeeds.

> On your startNotification listener we get always 8bytes, so it seems we get 4 responses with always 8 bytes.

Your peripheral must be sending this in 8 byte chunks. You can verify this by using an app like LightBlue Explore or nRF Connect. Subscribe to the notification and write the same data.

closed 2018-05-07 by don