don/cordova-plugin-ble-central #536: Observable and Promise

Domvel, 2018-02-13:
Is there a reason why return values are mixed? Observable and Promise. 
e.g. startNotify returns a Promise (with any type btw. It's an ArrayBuffer!). The read method returns a Promise (also any).

mburger81, 2018-02-14:
IMO read does not need an Observable because there is only ONE ArrayBuffer to return, so PROMISE is the right choice over using Observable which is used for startNotifiy which is success callback is called on every notification.

don, 2018-04-19:
This plugin uses callbacks and has optional and [optional API with promises](https://github.com/don/cordova-plugin-ble-central/blob/master/www/ble.js#L166). Most people see the Observable and Promises from the Ionic Wrapper https://ionicframework.com/docs/native/ble/?

If there's one success or one failure, a Promise is used. Some cases, like connect, return success when the connection succeeds and then later can call the failure callback if there is a disconnect. Promises can only be called once, so Observable is used for connect. Notifications call the success function 0 to infinite times, so they need an Observable too. Observables could be used everywhere but the API would be more confusing.

closed 2018-04-19 by don