don/cordova-plugin-ble-central #541: Can't find any device after several times of startScanWithOptions

lolosssss, 2018-03-02:
Hi,

I'm using the BLE plugin, and facing such an issue (only on Android):
After called `startScanWithOptions`  and `stopScan` for several times, it will find nothing. 
Relaunch the app or close and reopen the Bluetooth in system, it works again.

Here you can find the [demo](https://github.com/lolosssss/ionic-ble-demo)

I've tested on 
1. onePlus 5T, Android 7.1.1 
2. Samsung Galaxy S6, Android 7.0

Dev Env:
```
    @ionic/cli-utils  : 1.19.1
    ionic (Ionic CLI) : 3.19.1

System:

    Node : v8.9.4
    npm  : 5.6.0
    OS   : macOS High Sierra
```

Any idea?

VolodymyrAndrushchak, 2018-03-02:
@lolosssss 
I met the same problem on Android platform @don. 
I have the next problem - `startScanWithOptions` stops to run after 20min or 4 hours or 8hours. 

I had an idea to switch off and switch on the Bluetooth but it's not good idea (user's permission)
But in my app, I don't use `stopScan` and `startScanWithOptions` runs forever. However, I use `ble.connect()` function. I know that it's not a good idea but it's a real solution for me. And power consumption is acceptable!

lolosssss, 2018-03-02:
@VolodymyrAndrushchak 
Thank you for your advice.

In my case, the MAC address(or UUID) of the same BLE device is not fixed, and there is a string (like serial number) in advertising, so the scan function must be called before `ble.connect`

And I have built a native Android project, with the same issue. You can find it [here](https://github.com/lolosssss/ionic-ble-demo). 
It seems like it's not an ionic plugin relevant issue.

VolodymyrAndrushchak, 2018-03-02:
@lolosssss 
I got it.
Yes, I know about UUID - I had a project in production with iBeacons.

Then I can say only one thing - good luck!

closed 2018-04-27 by lolosssss