don/cordova-plugin-ble-central #542: scan() fails on 7.1.2

xkortex, 2018-03-05:
`cordova --version`
6.5.0
`cordova plugin ls`
cordova-plugin-ble-central 1.1.4 "BLE"
cordova-plugin-compat 1.2.0 "Compat"
cordova-plugin-whitelist 1.3.3 "Whitelist"

I have your fantastic BLE plugin working well on `LG G6` on `7.0` and `Moto Droid Maxx` on `4.4.4`. However, I am trying to deploy to a `LG V30+` on `7.1.2` and `ble.scan()` fails with no error in the chrome/inspect console. The target is a Adafruit Feather nRF52 Bluefruit LE - nRF52832 chipset, running custom firmware. I'm able to UART in real time on the G6 and the Moto no problem. The app fails to locate any devices on scan on the V30.

In `chrome://inspect`, I get this:

> ble.scan([], 3, echo, fail)
< undefined

(where echo and fail are just callbacks like echo = e => {console.log('success', e)})

Here is [paste of adb logcat](https://pastebin.com/gDPGxLyT).  

This appears to be the error that it is throwing: 

```
D BluetoothAdapter: startLeScan(): null
D BluetoothAdapter: STATE_ON
E BluetoothLeScanner: application registeration exception
E BluetoothLeScanner: android.os.DeadObjectException
E BluetoothLeScanner: 	at android.os.BinderProxy.transactNative(Native Method)
E BluetoothLeScanner: 	at android.os.BinderProxy.transact(Binder.java:615)
E BluetoothLeScanner: 	at android.bluetooth.IBluetoothGatt$Stub$Proxy.registerClient(IBluetoothGatt.java:877)
E BluetoothLeScanner: 	at android.bluetooth.le.BluetoothLeScanner$BleScanCallbackWrapper.startRegisteration(BluetoothLeScanner.java:304)
E BluetoothLeScanner: 	at android.bluetooth.le.BluetoothLeScanner.startScan(BluetoothLeScanner.java:193)
E BluetoothLeScanner: 	at android.bluetooth.le.BluetoothLeScanner.startScan(BluetoothLeScanner.java:113)
E BluetoothLeScanner: 	at android.bluetooth.BluetoothAdapter.startLeScan(BluetoothAdapter.java:2771)
E BluetoothLeScanner: 	at android.bluetooth.BluetoothAdapter.startLeScan(BluetoothAdapter.java:2687)
E BluetoothLeScanner: 	at com.megster.cordova.ble.central.BLECentralPlugin.findLowEnergyDevices(BLECentralPlugin.java:489)
E BluetoothLeScanner: 	at com.megster.cordova.ble.central.BLECentralPlugin.execute(BLECentralPlugin.java:141)
E BluetoothLeScanner: 	at org.apache.cordova.CordovaPlugin.execute(CordovaPlugin.java:117)
E BluetoothLeScanner: 	at org.apache.cordova.CordovaPlugin.execute(CordovaPlugin.java:98)
E BluetoothLeScanner: 	at org.apache.cordova.PluginManager.exec(PluginManager.java:132)
E BluetoothLeScanner: 	at org.apache.cordova.CordovaBridge.jsExec(CordovaBridge.java:57)
E BluetoothLeScanner: 	at org.apache.cordova.engine.SystemExposedJsApi.exec(SystemExposedJsApi.java:41)
E BluetoothLeScanner: 	at org.chromium.base.SystemMessageHandler.nativeDoRunLoopOnce(Native Method)
E BluetoothLeScanner: 	at org.chromium.base.SystemMessageHandler.handleMessage(SystemMessageHandler.java:9)
E BluetoothLeScanner: 	at android.os.Handler.dispatchMessage(Handler.java:102)
E BluetoothLeScanner: 	at android.os.Looper.loop(Looper.java:154)
E BluetoothLeScanner: 	at android.os.HandlerThread.run(HandlerThread.java:61)
D BLEPlugin: Stopping Scan
```
Let me know any other diagnostics I can run. It may take me some time to get back to you with other diagnostics, as it is the client's phone, so I have to pay them a visit (next city over) to plug into it.

xkortex, 2018-03-06:
Update: Power toggling the device's bluetooth allowed me to get the connection established on the phone. 

I *think* I can work around this interface by prompting the user to disable/enable using the API. Janky, but more workable than a dead connection.

don, 2018-04-25:
@xkortex I'm not sure there's a better solution. I think you just have a flaky Bluetooth stack on the V30+.Let me know if you see this behavior on more devices.

closed 2018-05-07 by don