don/cordova-plugin-ble-central #543: RSSI update on scan

mirkomancin, 2018-03-12:
Is it possible to update RSSI device info during scan without stop and restart scan? I read in the code that the list of peripherals is emptied before the scan (into `findLowEnergyDevices ` method). Is it possible to insert new option on scanWithOptions that select if or not empty this list?

VolodymyrAndrushchak, 2018-03-12:
@mirkomancin 

According to the RSSI. You can use `readRSSI` [method](https://github.com/don/cordova-plugin-ble-central#readrssi). Also, the scan method returns an updated RSSI level of BLE devices.

In `scanWIthOption` you can set empty array or set UUID for scanning only specified devices.

mirkomancin, 2018-03-12:
readRSSI works only if you are connected to device. Else, if I try to rescan and I try to connect to a device not yet found by the scan fails because it is not in the list of peripherals. I would have a method to refresh rssi of alla peripherals founded during scanning operation without connect to them

VolodymyrAndrushchak, 2018-03-12:
@mirkomancin 
Okay
In my case, I use the `scanWithOption` and I get an RSSI from this method. I don't know where you use this plugin (I use on Android and iOS platform) but in my case, it works

mirkomancin, 2018-03-12:
But the problem remain. Because if I have device at distance of 15m,
latency between scan and discovery is of few seconds. And if I try to
connect during this time, I receive an error because list not contain my
peripheral

VolodymyrAndrushchak, 2018-03-12:
@markterrill 
which platform do you use?

mirkomancin, 2018-03-12:
Android

VolodymyrAndrushchak, 2018-03-12:
@mirkomancin 
I also had the same problem on Android. I don't think that it's possible to configure from a cordova level. 
I used  `connect` and `readRSSI` methods on Android. Yes, I also had the same latency. It also depends on the configuration of your ble devices. These devices send a signal periodically and you can reprogram you ble. 
I don't know if you use this plugin on iOS, but on this platform(iOS) it works perfectly.

mirkomancin, 2018-03-12:
It is sufficient It is sufficient to don't empty the list of periherals
when starting the scan with a flag to be added in the options. I don't know
how to recompile this feature, else I have already write it into my code.

closed 2018-05-07 by don