don/cordova-plugin-ble-central #544: ble.startNotification doesn't work

NBrandacher, 2018-03-26:
Starting situation:
I'm currently building an app that is communicating with an ESP32 over BLE. The ESP32 is already set up as a GATT-Server, sending the required data once per second. I've tested this using the app "LightBlue" where I connected to the ESP32 and listened for notifications. So far I've built an app which is able to scan for ble-devices and connect to them in order to get the information of the peripheral.

So now to get the data, that the ESP32 is already sending, I wanted to use the ble.startNotification function which apparently doesn't work. When the function is called it jumps to the failure callback-function even though the characteristic_uuid and the service_uuid should be correct. 

Any ideas what could be wrong, maybe someone has already built apps for communication with an ESP32.

jayeshkorat18, 2018-04-02:
After successfully connect i try to ble.startNotifucation but doesn't respond success of failed

don, 2018-04-19:
ble.startNotification works, there's likely an error in your code. 

Take a look at the [heartrate example](https://github.com/don/cordova-plugin-ble-central/blob/master/examples/heartrate/www/js/index.js) for how to notifications with this plugin.

chicag0, 2018-04-30:
I'm also having issues with this.  I can subscribe to events using LightBlue so I know they are firing, but I do not get success or failure messages when using this

don, 2018-05-01:
@chicag0 this is likely a problem in your app, can you provide a simple example that demonstrates the problem?

LeHoang93, 2018-05-09:
Hi i got the same Problem with the startNotification() function.
I have two iTag Buttons from different manufacturers, with the same Button case.

### First Button:

[image]

[image]

[image]

The first button works fine, i can connect to the button and register the notification.

### Second Button:

[image]

[image]

The second button does not work right, i can connect to the device, but can't register to be notified.
I think the reason is the missing descriptor.

```
console.error: Unhandled Promise rejection: Set notification failed for 0000ffe1-0000-1000-8000-00805f9b34fb 
            ; Zone: <root> ; Task: Promise.then ; Value: Set notification failed for 
            0000ffe1-0000-1000-8000-00805f9b34fb 
```
That's the error message then i call startNotification

@don Do you have any idea how i can fix it? I would appreciate your help.

Sorry for my bad english.

closed 2018-05-20 by don