don/cordova-plugin-ble-central #545: Cannot find any devices

RitterD, 2018-03-28:
Hello,

first of all I'm completely new on github and english is not my motherlanguage so please forgive me my grammer and spelling misstakes.

My name is Daniel Ritter and I'm trying to build a smartphone app for my bachelor thesis in june also I'm an engineer and not really a programmer....

Nevertheless I'm trying to implement your ble plugin with the documentation you deliverd.

I tryed to your scan function with a button but I cannot find any devices...

`<button onclick="ble.scan([], 10, function(device) {
    console.log(JSON.stringify(device));
}, failure);"
 name="scan">Start scanning</button>`

I'm supposed to find some peripheral devices with this right? But where are they shown I really dont get it....
I tryed your ble-scan-master but couldnt find any devices also..

Here is a link to download my project folder:
https://www.dropbox.com/s/b80cw8rio5phj2k/myapp1.zip?dl=0

Please give me a hint

All the best
Daniel Ritter

VolodymyrAndrushchak, 2018-03-29:
Hello @RitterD 

First of all, it isn't a good idea to use js directly in HTML. 

Be sure, that your BLE devices are visible. You can check it by using next Apps:
- nRF Toolbox 
- BLE Scanner & Recorder
- etc - you can find a lot in Stores

If you know which BlE devices you scan, you have to use `scanWithOptions()` and specify UUID of your ble devices. 

This plugin works. Check if you turn on Bluetooth on your mobile phone.
If you run on Android check your Android version - at least 4.3 should be.

RitterD, 2018-03-29:
Hi volodymyr,

thank you very mutch for the quick answer I really appreciate =)

> First of all, it isn't a good idea to use js directly in HTML
yes I know, but it was my first app and I just want to try some stuff out.

I'm sure the plugin works, but I think I didn't implement it correctly....
Wait a second, did you try it with my code and it worked?!

VolodymyrAndrushchak, 2018-04-03:
@RitterD 
I haven't tried it

is it cordova or ionic project?

don, 2018-04-19:
@RitterD Try this example to scan for any devices https://github.com/don/ITP-BluetoothLE/tree/master/phonegap/scan 

Compare the results using [LightBlue Explorer](https://itunes.apple.com/us/app/lightblue-explorer/id557428110?mt=8)

There's also an Ionic version https://github.com/don/ionic-ble-examples/tree/master/scan

RitterD, 2018-04-20:
@VolodymyrAndrushchak it's a cordova project, but it i changed everything, some stuff is still written in german, sorry for that...

here the link to my current version:
https://www.dropbox.com/s/obrh667e2tx7zum/myapp%20V1.rar?dl=0

@don thank you very mutch, ill try it out next week

closed 2018-06-07 by don