don/cordova-plugin-ble-central #547: Enable bluetooth on iOS?

chicag0, 2018-04-05:
Hello,

Just reading the docs and it says "enable" is for Android only, and also that "showBluetoothSettings" is Android only as well --- what is the suggested way to detect iOS and get the user to enable Bluetooth?

chicag0, 2018-04-09:
Hi @don do you have any ideas?  I cant seem to get isEnabled to work on iOS either, it always thinks it is disabled...

VolodymyrAndrushchak, 2018-04-10:
@chicag0 

Hello

I use `ble.isEnabled()` function and it works perfectly.

But you must specify the item in the Plist - 'Privacy - Bluetooth Peripheral Usage Description' and add a description to this item

chicag0, 2018-04-10:
@VolodymyrAndrushchak 

Thanks for the tip -- I'll make sure I have that item in the Plist

VolodymyrAndrushchak, 2018-04-10:
@chicag0 

Okay, what happens when you call `startScanWithOptions`?

chicag0, 2018-04-10:
@VolodymyrAndrushchak 

Using the sample code I am getting the failure response.  I think that is expected given that 'isEnabled' is also returning false -- any other ideas?   Does the format of my plist keys look correct?

VolodymyrAndrushchak, 2018-04-10:
@chicag0 

Where have you tried to run your app? I mean iOS version and iPhone. 
I hope that you have tried on a real device)

chicag0, 2018-04-10:
@VolodymyrAndrushchak 

Yes, I have an iPad and an iPhone that I am testing on.

VolodymyrAndrushchak, 2018-04-10:
@chicag0 
Can you write iPhone type and iOS version?

chicag0, 2018-04-18:
@VolodymyrAndrushchak 

Testing on both an older iPad2 running IOS 9.3.5 as well as a newer iPhone 7 running 11.3

 ble.isEnabled() is always returning false on IOS even when enabled.  Do my usage descriptions above look correct?

chicag0, 2018-04-19:
@VolodymyrAndrushchak 

Well I just went out and purchased a brand new iPad, I think my iPad 2 is just too old to support Bluetooth 4.

Anyways, now it is kind of working!  But I noticed that when running startStateNotifications I dont get any notifications, where on Android I can run it and it will pick up the current state.

Is there another way on iOS to do this

VolodymyrAndrushchak, 2018-04-19:
@chicag0 
I am glad to hear that something already is working)

It seems that iOS has changed the policy or even API for bluetooth enabling. but I can't say exactly info.   I will try to run it on iOS on next Monday and will give you the response!

don, 2018-04-19:
Unfortunately iOS doesn't provide a way to enable Bluetooth programmatically, so ble.enable is Android only.

iOS should return state notifications with startStateNotifications. Check for JavaScript errors causing problems. (Looking at the console output on your phone using Safari's debug menu on your computer.)

don, 2018-04-23:
In my apps, I'm always handling state notifications regardless of the page the user is on. This allows me to update the UI with the bluetooth state and depending on what's happening show a dialog. If for some reason there user doesn't care about Bluetooth state, the event handler will ignore the event.

chicag0, 2018-04-23:
@don 

Thanks.   I'm going to close this issue for now.

closed 2018-04-23 by chicag0

JhonnyJason, 2018-06-26:
Just to be sure - this means that there is no way to enable bluetooth from inside the app?

The way to go is to check if 
```javascript
ble.isEnabled(onBLEOn, onBLEOff)
```
And if it is not use
```javascript
ble.startStateChangeNotifications(stateChange)
```
to recognize that we became connected.
??

Thanx :-)

don, 2018-06-29:
There is not way to enable Bluetooth settings on iOS.

The code to do it is in https://github.com/don/cordova-plugin-ble-central/pull/591 but it was removed because it causes apps to get rejected from the store.

ble.startChangeNotifications is generally better than ble.isEnabled, but you might need both. Use ble.isEnabled to know if you need to prompt the user and use change notifications to know when it is enabled.

stefanotauriello, 2020-05-29:
> @chicag0
> 
> Hello
> 
> I use `ble.isEnabled()` function and it works perfectly.
> 
> But you must specify the item in the Plist - 'Privacy - Bluetooth Peripheral Usage Description' and add a description to this item

@VolodymyrAndrushchak ble.isEnabled() return null.... why?